yaqwsx/KiKit #571: 'SwigPyObject' object has no attribute 'FootprintLoad'

bradanlane, 2023-07-24:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

1.3.0+25.g508cc19

### KiCAD version

7.0.5

### Operating system

Windows 11

### Description

attempting to create a simple 1x1 panel. It's not clear if the issue is an oddity with the PCB or a bug in KiKit. The PCB passed DRC.
```
An error occurred: 'SwigPyObject' object has no attribute 'FootprintLoad'
No output files produced
Traceback (most recent call last):
  File "C:\Users\labmonitor\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\panelize_ui.py", line 217, in panelize
    doPanelization(input, output, preset, plugin)
  File "C:\Users\labmonitor\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\panelize_ui.py", line 278, in doPanelization
    ki.buildTooling(preset, panel)
  File "C:\Users\labmonitor\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\panelize_ui_impl.py", line 507, in buildTooling
    panel.addCornerTooling(4, hoffset, voffset, diameter, paste)
  File "C:\Users\labmonitor\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\panelize.py", line 1477, in addCornerTooling
    self.addNPTHole(pos, diameter, paste)
  File "C:\Users\labmonitor\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kikit\panelize.py", line 1401, in addNPTHole
    footprint = pcbnew.FootprintLoad(KIKIT_LIB, "NPTH")
  File "C:\Program Files\KiCad\bin\Lib\site-packages\pcbnew.py", line 18579, in FootprintLoad
    return plug.FootprintLoad(libname,name,preserveUUID)
AttributeError: 'SwigPyObject' object has no attribute 'FootprintLoad'
```

### Steps to Reproduce

``` JSON
{
    "layout": {
        "type": "grid",
        "rows": 1,
        "cols": 1,
        "hspace": "2mm",
        "vspace": "2mm"
    },
    "source": {
        "type": "auto",
        "tolerance": "2mm"
    },
    "tabs": {
        "type": "corner",
		"width": "4mm",
        "source": "none"
    },
	"cuts": {
		"type": "mousebites",
		"drill": "0.5mm",
		"spacing": "0.75mm",
		"offset": "-0.35mm",
		"prolong": "-0.4mm"
	},
    "framing": {
        "type": "tightframe",
		"hspace": "5mm",
		"vspace": "5mm",
		"slotwidth": "2.5mm",
        "width": "2.5mm"
    },
	"tooling": {
		"type": "4hole",
		"hoffset": "3.5mm",
		"voffset": "3.5mm",
		"size": "1.8mm"
	},
	"fiducials": {
		"type": "4fid",
		"paste": true,
		"hoffset": "13.5mm",
		"voffset": "3.5mm",
		"coppersize": "1mm",
		"opening": "1.5mm"
	},
	"post": {
		"type": "auto",
		"millradius": "1mm",
		"origin": "bl",
        "copperFill": false
	}
}
```

yaqwsx, 2023-07-24:
Based on passing test suite and test performed in my setup, I suspect problem in your or Windows installation of KiCAD.

Could you run Python interpreter in the KiCAD Command prompt and put the following commands in there and share the output?

```
>>> import pcbnew
>>> pcbnew
>>> pcbnew.FootprintLoad
```

and then in a new interpreter:
```
>>> from pcbnewTransition import pcbnew
>>> pcbnew
>>> pcbnew.FootprintLoad
```

bradanlane, 2023-07-24:
Not sure what the results are meant to tell me but here they are ...

```
Python 3.9.16 (main, May 18 2023, 07:48:16) [MSC v.1936 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pcbnew
>>> pcbnew
<module 'pcbnew' from 'C:\\Program Files\\KiCad\\bin\\Lib\\site-packages\\pcbnew.py'>
>>> pcbnew.FootprintLoad
<function FootprintLoad at 0x0000024B2A13BAF0>
```
... and ...
```
Python 3.9.16 (main, May 18 2023, 07:48:16) [MSC v.1936 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pcbnewTransition import pcbnew
>>> pcbnew
<module 'pcbnewTransition.pcbnew' from 'C:\\Users\\labmonitor\\Documents\\KiCad\\7.0\\3rdparty\\Python39\\site-packages\\pcbnewTransition\\pcbnew.py'>
>>> pcbnew.FootprintLoad
<function FootprintLoad at 0x00000266E52DF040>
```

yaqwsx, 2023-07-24:
The output looks fine. In the case it works on other PCB, would you mind sharing the PCB that triggers the behavior? You can also do it non-publically via e-mail: email@honzamrazek.cz.

bradanlane, 2023-07-24:
emailing

It has a number of customer footprints from a global library so opening it will likely have errors.

bradanlane, 2023-07-25:
As a general comment, when I've had errors from KiKit, I can not determine if the error is related to the JSON settings, a specific component on the PCB, or something else.

If the error is related to a specific component (or trace, text, etc), would it be possible to include the description, ID, or other identifying attribute in the error message? 

This would often reduce the diagnostics effort.

bradanlane, 2023-07-25:
additional information:
I uninstalled KiKit and installed.
It is generating a large number of messages similar to `swig/python detected a memory leak`. Here are a few examples _(not a full output; only a few individual lines)_:
```
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
```

The PCB has an irregular edge cut comprised of lots of small line segments. The other PCB that has no problem for KiKit has a simple circle edge cut.

bradanlane, 2023-07-25:
The issue is with the complex edge cut!

yaqwsx, 2023-07-25:
I just tested the board you sent me, and it works fine on the latest master (that is `kikit, version 1.3.0+39.g07c707d
`). Could you test it? The error you observed should not have anything to do with it. Internal KiCAD function fails. The whole thing is strange.

bradanlane, 2023-07-25:
let me go find the instructions for installing KiKit from the repo.

bradanlane, 2023-07-25:
here is the "project" consisting of just the edge cut ...
[kikit_issue_571.zip](https://github.com/yaqwsx/KiKit/files/12163228/kikit_issue_571.zip)

bradanlane, 2023-07-25:
things seem to be getting weirder with each attempt to work around.

I simplified the edge cut and can avoid the error. However,now KiKit is rotating the edge cut 180 degrees (or flipping it vertically). The edge cut is part of a footprint and that footprint is on the back of the PCB.

bradanlane, 2023-09-03:
The error has returned when I use `KiCad 7.0.7` and `KiKit version 1.3.0+42.g63301c2`.

bradanlane, 2023-09-05:
here is the latest project with just the end cut. It reproduces the error with `kikit, version 1.3.0+42.g63301c2`
[kikit_issue_571_20230905.zip](https://github.com/yaqwsx/KiKit/files/12520769/kikit_issue_571_20230905.zip)

bradanlane, 2023-09-05:
I am now very confused ...

1. I deleted all of KiCad, Python, pip, and KiKit.
2. I did a clean install of KiCad 7.0.5 and KiKit version 1.3.0+39.g07c707d. This is the configuration which matches 5-Jul-2023 above.
3. I downloaded the `kikit_issue_571.zip` from 5-Jul-2023.
4. I get the error. 

Somehow, we got the error to go away with this configuration.
I can not identify what is different.

bradanlane, 2023-09-05:
On Windows, do I need to "run as Administrator" for the install step and/or when running the kikit command?

bradanlane, 2023-09-05:
If you can suggest of a workaround, I'm happy to try just about anything at this point.

yaqwsx, 2024-03-10:
After 2 new releases of KiKit and the release of KiCAD 8 - is the problem solved or not? I struggled a lot with reproducing it and I still suspect a packaging issue in KiCAD rather than a bug in KiKit.

bradanlane, 2024-03-10:
I have not been able to upgrade KiCAD since 7.0.7 because later 7.- released broke multiple things related to edge cuts.

I cannot move to 8.x until my current designs are done and shipped because I can't not afford the regression cycles.

If you want to close this, then go ahead. If this issue reappears after I move to 8.x I can either reopen it or create a new issue.

closed 2024-03-13 by yaqwsx