yaqwsx/KiKit #572: Footprint with edgecut placed on back causes KiKit to flip edcut vertically

bradanlane, 2023-07-25:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0+39.g07c707d

### KiCAD version

7.0.5

### Operating system

Windows 11

### Description

If the PCB edge cut is part of a footprint and that footprint is placed on the back of the PCB, then KiKit flips the edge cut vertically. It should flip horizontally.

### Steps to Reproduce

- create a footprint which contains an irregular shaped footprint. _(the irregularity to to make it easy to identify the orientation of the footprint and the edge cut)_
- create a PCB with this footprint
- position the footprint on the back of the PCB _(this will flip the irregular edge cut left/right)_
- generate a 1x1 panel

The edge cut on the panel will have been flipped top/bottom.

Source PCB:
[image]

KiKit Result:
[image]

yaqwsx, 2023-08-17:
I tried to reproduce your issue - I created the following project:
[example.zip](https://github.com/yaqwsx/KiKit/files/12371283/example.zip)
[image]

After running `kikit panelize ~/projects/kikit/repro/mfootprint/mfootprint.kicad_pcb panel.kicad_pcb` I get:
[image]

Which looks just right. Tested on `kikit, version 1.3.0+42.g63301c2`. Could you try the upstream version? If it doesn't help, could you share the project (problematic footprint and outline should be sufficient).

bradanlane, 2023-08-17:
I will test with the upstream. It's odd but I thought I added a zip of a simple project to reproduce the issue. If it still persists I will insure test file appears here.

bradanlane, 2023-08-17:
I tested with the current upstream but used our footprint and am still able to reproduce. attached file.
[mfootprint_bradan.zip](https://github.com/yaqwsx/KiKit/files/12371441/mfootprint_bradan.zip)

yaqwsx, 2023-08-18:
Ah, the difference is in whether you use lines or polygon. With polygons, there is something strange happening inside KiCAD that causes the problem. I wasn't able to trace it yet. As a work-around before we find fix I suggest converting the polyline outline to lines.

yaqwsx, 2024-03-10:
I just retested this issue with KiCAD 8 and it seems to me that the problem was fixed. Could you confirm it (after a long time I am unsure if I executed the steps for reproduction correctly).

bradanlane, 2024-03-10:
I have not been able to upgrade KiCAD since 7.0.7 because later 7.- released broke multiple things related to edge cuts.

I cannot move to 8.x until my current designs are done and shipped because I can't not afford the regression cycles.

If you want to close this, then go ahead. If this issue reappears after I move to 8.x I can either reopen it or create a new issue.

yaqwsx, 2024-03-10:
Could you elaborate more on " 7.- released broke multiple things related to edge cuts."? Why I am asking: I observed a bug in KiCAD's handling of edges in footprints. If 7.0.8 broke something, it might be related to fixing it. Knowing the details could help us trace the root cause.

bradanlane, 2024-03-10:
I encountered 2 issues (one started with 7.0.8 and another appeared with 7.0.9).

Edge cut was no longer considered closed but no error appeared. Filled zones stop filling.

Irregular openings within edge cuts started to generate hundreds of DRC errors related to net segments and bias being outside edge cuts (even though the board was unchanged from 7.0.7).