yaqwsx/KiKit #574: Shapely >= 2.0 Support

r0hit05, 2023-07-31:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

1.3.0

### KiCAD version

N/A

### Operating system

N/A

### Description

Not a bug, but if you ever bump up the required version for `shapely` to `>= 2.0`, please let us know. 

### Steps to Reproduce

N/A

yaqwsx, 2023-07-31:
This is not planned as I plan to migrate to completely different geometric engine.

hvraven, 2023-12-08:
Is there a plan when this change will happen? It appears that shapely 1.8.5.post1 (the latest <2) is not compatible with Python 3.12 due to removal of long deprecated code in pkgutil.

yaqwsx, 2024-01-06:
It seems that Shapely 2.0.2 removes the inconsistency in behavior that caused problems for KiKit. So I allowed Shapely 2 for KiKit in 3427a8d. If you run into any problems (e.g., warnings from Shapely printed on stderr or broken geometry), please, let me know.

zapta, 2024-01-06:
I was referred from #625, installation of kikit still fails. Windows log below:

```bash
~$ python --version
Python 3.12.0
~$ pip --version
pip 23.3.2 from C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip (python 3.12)
~$ pip show kikit
WARNING: Package(s) not found: kikit
~$ pip install kikit
Collecting kikit
  Using cached KiKit-1.4.0-py3-none-any.whl.metadata (5.7 kB)
Collecting click>=7.1 (from kikit)
  Using cached click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
Collecting commentjson>=0.9 (from kikit)
  Using cached commentjson-0.9.0.tar.gz (8.7 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting markdown2>=2.4 (from kikit)
  Using cached markdown2-2.4.12-py2.py3-none-any.whl.metadata (2.0 kB)
Requirement already satisfied: numpy in c:\users\user\appdata\local\programs\python\python312\lib\site-packages (from kikit) (1.26.0)
Collecting pcbnewTransition<=0.4,>=0.3.4 (from kikit)
  Using cached pcbnewTransition-0.3.4-py3-none-any.whl (6.7 kB)
Collecting pybars3>=0.9 (from kikit)
  Using cached pybars3-0.9.7.tar.gz (29 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting shapely<2.0,>=1.7 (from kikit)
  Using cached Shapely-1.8.5.post1.tar.gz (200 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'error'
  error: subprocess-exited-with-error

  Getting requirements to build wheel did not run successfully.
  exit code: 1

  [33 lines of output]
  Traceback (most recent call last):
    File "C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
      main()
    File "C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 112, in get_requires_for_build_wheel
      backend = _build_backend()
                ^^^^^^^^^^^^^^^^
    File "C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 77, in _build_backend
      obj = import_module(mod_path)
            ^^^^^^^^^^^^^^^^^^^^^^^
    File "C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\importlib\__init__.py", line 90, in import_module
      return _bootstrap._gcd_import(name[level:], package, level)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "<frozen importlib._bootstrap>", line 1381, in _gcd_import
    File "<frozen importlib._bootstrap>", line 1354, in _find_and_load
    File "<frozen importlib._bootstrap>", line 1304, in _find_and_load_unlocked
    File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
    File "<frozen importlib._bootstrap>", line 1381, in _gcd_import
    File "<frozen importlib._bootstrap>", line 1354, in _find_and_load
    File "<frozen importlib._bootstrap>", line 1325, in _find_and_load_unlocked
    File "<frozen importlib._bootstrap>", line 929, in _load_unlocked
    File "<frozen importlib._bootstrap_external>", line 994, in exec_module
    File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
    File "C:\Users\user\AppData\Local\Temp\pip-build-env-8hi25y1h\overlay\Lib\site-packages\setuptools\__init__.py", line 16, in <module>
      import setuptools.version
    File "C:\Users\user\AppData\Local\Temp\pip-build-env-8hi25y1h\overlay\Lib\site-packages\setuptools\version.py", line 1, in <module>
      import pkg_resources
    File "C:\Users\user\AppData\Local\Temp\pip-build-env-8hi25y1h\overlay\Lib\site-packages\pkg_resources\__init__.py", line 2191, in <module>
      register_finder(pkgutil.ImpImporter, find_on_path)
                      ^^^^^^^^^^^^^^^^^^^
  AttributeError: module 'pkgutil' has no attribute 'ImpImporter'. Did you mean: 'zipimporter'?
  [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

Getting requirements to build wheel did not run successfully.
exit code: 1

See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
~$

```

yaqwsx, 2024-01-06:
@zapta: the changes are not released as we are not 100% they work. To test them, you have to install the upstream version, not the released one: https://yaqwsx.github.io/KiKit/v1.4/installation/upgrading/

zapta, 2024-01-06:
Thanks Jan.  I ran the command line below and it failed. Please let me know
if I should run something else.

Also, please post here when the fix will be pushed out.

```
~$ pip install ***@***.***
Collecting ***@***.***
  Cloning https://github.com/yaqwsx/KiKit (to revision master) to
c:\users\user\appdata\local\temp\pip-req-build-lwhsf0br
  Running command git clone --filter=blob:none --quiet
https://github.com/yaqwsx/KiKit
'C:\Users\user\AppData\Local\Temp\pip-req-build-lwhsf0br'
  Resolved https://github.com/yaqwsx/KiKit to commit
db1861574d7f43a055271b895a826c4234007b2b
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'error'
  error: subprocess-exited-with-error

  Getting requirements to build wheel did not run successfully.
  exit code: 1

  [30 lines of output]

C:\Users\user\AppData\Local\Temp\pip-req-build-lwhsf0br\versioneer.py:421:
SyntaxWarning: invalid escape sequence '\s'
    LONG_VERSION_PY['git'] = '''

====================================================================================================

====================================================================================================
  ** Cannot install KiKit**
  No Pcbnew Python module found.
  Please make sure that you use KiCAD command prompt, not the standard
Command Prompt or Power Shell
  See
https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md#installation-on-windows

====================================================================================================

====================================================================================================
  Traceback (most recent call last):
    File
"C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py",
line 353, in <module>
      main()
    File
"C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py",
line 335, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File
"C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py",
line 118, in get_requires_for_build_wheel
      return hook(config_settings)
             ^^^^^^^^^^^^^^^^^^^^^
    File
"C:\Users\user\AppData\Local\Temp\pip-build-env-hj9y2web\overlay\Lib\site-packages\setuptools\build_meta.py",
line 325, in get_requires_for_build_wheel
      return self._get_build_requires(config_settings,
requirements=['wheel'])

 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File
"C:\Users\user\AppData\Local\Temp\pip-build-env-hj9y2web\overlay\Lib\site-packages\setuptools\build_meta.py",
line 295, in _get_build_requires
      self.run_setup()
    File
"C:\Users\user\AppData\Local\Temp\pip-build-env-hj9y2web\overlay\Lib\site-packages\setuptools\build_meta.py",
line 480, in run_setup
      super(_BuildMetaLegacyBackend,
self).run_setup(setup_script=setup_script)
    File
"C:\Users\user\AppData\Local\Temp\pip-build-env-hj9y2web\overlay\Lib\site-packages\setuptools\build_meta.py",
line 311, in run_setup
      exec(code, locals())
    File "<string>", line 35, in <module>
  RuntimeError: Cannot install KiKit, see error message above
  [end of output]

  note: This error originates from a subprocess, and is likely not a
problem with pip.
error: subprocess-exited-with-error

Getting requirements to build wheel did not run successfully.
exit code: 1

See above for output.

note: This error originates from a subprocess, and is likely not a problem
with pip.
```

yaqwsx, 2024-01-30:
Strangely, your OS reports as Windows, not Linux. Nevertheless, the problem is that the installation cannot locate the `pcbnew` module that is provided by KiCAD. Make sure that when you enter `import pcbnew` into Python interpreter, it doesn't fail.

yaqwsx, 2024-02-20:
There seem to be no more problems reported with Shapely 2 for a over a month, closing.

closed 2024-02-20 by yaqwsx