yaqwsx/KiKit #577: KiKit CLI command broken

JohK, 2023-08-01:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

python -m kikit.ui, version 1.3.0

### KiCAD version

7.0.6-0, release build

### Operating system

macos 12.6.5

### Description

KiKit CLI command output in gui is broken and cannot simply copied, because special characters are used that do not correspond to the actual commands on CLI.

Example:
```
kikit panelize \
    —layout ‘hbackbone: 9.5mm; vbackbone: 10mm’ \
[...]
```
1. it should be "--" not "—" to start the option off,
2. it should be a simple tick instead of a forward or backward tick ‘ -> ' and ’ -> '

### Steps to Reproduce

1. Run the gui
2. generate a panel
3. look at the generated CLI output
4. copy paste to shell

yaqwsx, 2023-08-01:
This has to be some kind of MacOS "fancy" thing or "font" fancy thing where it replaces `--` with `–`. Please, check the behavior of your default monospace font. I am not sure if there is anything I can do about it.

JohK, 2023-08-01:
When starting kicad I actually get some error message about a failure to load Courrier. This started with kicad 7.0.6.

Afaik for macos there is no default font. Maybe something WX does?

JohK, 2023-08-01:
Additional thought: If it was the font, it would be just the representation... the shell would be ignorant of the visual representation and just execute it. So something needs to convert the text when showing it in the text widget.

yaqwsx, 2023-08-02:
Since I don't own Mac, I cannot reproduce a trace the issue. I can only make wild guesses. I can assure you that we don't output `–` instead of `--` anywhere in the KiKit codebase. I suspect the macOS widget for text or your font is rendering ligatures.

JohK, 2023-08-02:
Ligatures do not change the underlying strings, it's just a different visual representation of two (or more) chars... so the shell would not care. But here the characters are actually exchanged.

I had a quick trip to the wx documentation for the TextCtrl() that is used to show the text. It may be actually rich text that is shown.

BUT, it is fixed for me now, after updating to the upstream version (1.3.0+40.g44e6e22) for another issue. I now get the text and can copy/paste it as expected.

closed 2023-08-05 by yaqwsx