yaqwsx/KiKit #579: fab export fails

JohK, 2023-08-01:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

1.3.0

### KiCAD version

7.0.6

### Operating system

macos 12.6.5

### Description

```
kikit fab jlcpcb --no-drc --schematic myproj.kicad_sch --debug myproj-framed.kicad_pcb production/test
[...]
File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pcbnew.py", line 7827, in PlotLayer
    return _pcbnew.PLOT_CONTROLLER_PlotLayer(self)
SystemError: <built-in function PLOT_CONTROLLER_PlotLayer> returned a result with an error set
```

For another project it worked. What features in the project am I looking for that may cause this issue?

### Steps to Reproduce

Export this framed board for production.

yaqwsx, 2023-08-02:
Please, try the upstream version. This should already be addressed by 508cc19261444eb22ad924d528999cfb123db8f6.

JohK, 2023-08-02:
Works indeed.

closed 2023-08-02 by JohK