yaqwsx/KiKit #580: Fab export for multi Unit Symbols with multiple footprints

JohK, 2023-08-02:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

I have a Raspberry CM4 multi-unit symbol that consists of 6 units:
A: DF40C-100DS-0.4V (100pin Hirose connector)
B: DF40C-100DS-0.4V (100pin Hirose connector)
C: mounting hole with Würth solder in Threaded spacer
D: idem
E: idem
F: idem

It is a single footprint that consists of multiple parts soldered into the correct position. The fields are shared between the units of the symbol. Currently I don't think these symbols/footprints are supported by kikit fab jlcpcb export for assembly (BOM and POS).

Proposed solution:
With a Module M1 consisting of the units mentioned above, all the already supported features should work of fields that are named *_UNIT_X.

LCSC_UNIT_A -> Cxxxxx
LCSC_UNIT_B -> Cxxx
LCSC_UNIT_C -> ...

JLCPCB_CORRECTION_UNIT_A 0;0;180
JLCPCB_CORRECTION_UNIT_C 0;0;90

The generated reference could be composed of the Footprint reference + the unit letter. In this case M1A, M1B, M1C, etc

yaqwsx, 2023-08-03:
I believe this is the same functionality as in #118. The original idea was to have a set of "rewrite rules" that would split a single footprint into multiple assembly items.

JohK, 2023-08-03:
Yes, that's the same idea.