yaqwsx/KiKit #584: kicad 7 не закрывается

mikhailmihalkov, 2023-08-25:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

1.3.0

### KiCAD version

7.0.7

### Operating system

win 10/64

### Description

After installing the plugin both from the directory and through pip kicad after closing it continues to hang in the processes after removing the plugin closes well checked on two different machines.

### Steps to Reproduce

Install the plugin in any way start kicad 7.0.7 open the board file close everything start the task manager kicad continues to work

CRImier, 2023-10-02:
Pretty sure this is a duplicate of #590 . I'm having the same issue - try updating to 1.4.0! `pip install -U kikit`. I'll test 1.4.0 some and see if it has disappeared.

yaqwsx, 2023-10-13:
Please try it with 1.4.0 - there were changes related to closing dialogs that could resolve this. At the moment, I cannot reproduce this issue on Ubuntu 22.04.

yaqwsx, 2023-10-21:
Resolved, see #590 for details.

closed 2023-10-21 by yaqwsx