yaqwsx/KiKit #586: kikit installation error in windows

Baskaran-eee-83, 2023-09-07:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

1.3

### KiCAD version

7.0

### Operating system

Windows 10

### Description

C:\Users\Bas\Documents\KiCad\7.0>pip install kikit
Error in sitecustomize; set PYTHONVERBOSE for traceback:
AssertionError: SRE module mismatch
Could not import runpy module
Traceback (most recent call last):
  File "<frozen runpy>", line 15, in <module>
  File "<frozen importlib.util>", line 2, in <module>
ModuleNotFoundError: No module named 'importlib._abc'

when i try to install it show this error , pl guideme.

### Steps to Reproduce

just install via kicad terminal

yaqwsx, 2023-10-13:
This seems like a corrupted Python installation. We have already seen cases when anti-virus software decimated the Python interpreter packed with KiCAD:

yaqwsx, 2023-11-12:
One last thing; please make sure that you run the commands inside `KiCAD Command Prompt`, not just `Command Prompt`. Nevertheless, this isn't a problem in KiKit so closing this issue for the moment.

closed 2023-11-12 by yaqwsx