yaqwsx/KiKit #590: kitkit prevents proper shutdown of kicad on windows

Hoernchen, 2023-09-15:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.3.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0

### KiCAD version

7.0.7

### Operating system

win11 x64

### Description

kikit prevents kicad from shutting down after opening a pcb file - there is nothing special about the pcb, just installing but not using kikit from the package manager is enough.

### Steps to Reproduce

1) install kikit python package using the kicad command prompt
2) install kikit using the pcm (1.1.0 is still the newest version available?)
3) start kicad, open pcb
- opening is enough, no need to do anything else
4) quit kicad
-> kicad.exe keep running in the background forever.

fix: uninstall kikit using the pcm.

Opening a pcb file is required for the hang, this does not happen with eeschema.

CRImier, 2023-10-02:
I've been experiencing this on Ubuntu! Didn't think it could've been KiKit. And yeah, this happens exclusively with `kicad`/`pcbnew` executables - `eeschema` closes fine! It also has a memory leak in the background, which is how I even found out about this; ngl, it's a real annoying bug! On the upside, I've just updated to 1.4.0 and will see if this is resolved - can't reproduce right now, but it doesn't always happen, either.

yaqwsx, 2023-10-13:
Please try it with 1.4.0 - there were changes related to closing dialogs that could resolve this. At the moment, I cannot reproduce this issue on Ubuntu 22.04.

CRImier, 2023-10-20:
Yep, same, in these three weeks' time, can no longer reproduce after updating to 1.4.0! Think this issue is resolved.

closed 2023-10-21 by yaqwsx

Hoernchen, 2023-10-21:
Yes, I can confirm that it appears to be fixed on Windows as well, thanks!