yaqwsx/KiKit #593: Unexpected Panel Cutout Behavior

gcormier, 2023-10-01:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

1.4.0

### KiCAD version

7.0.8

### Operating system

Linux and Windows 11

### Description

Note - I am getting the same output on my local Windows machine, and my Ubuntu GitHub (linux) pipeline.

Here is my board
[image]

**Note in all cases, the top corners don't seem to be affected by the framing `fillet: 1mm` parameter.**

Changing the value of the `cutout` parameter in `tabs:`

## 1mm (default)
2 weird cutouts on an angle
[image]
[image]

## 1.5mm
all 4 weird cutouts
[image]

## 2mm
Slot?
[image]

## Notes
I've also tried to make this center cutout without a radius.
[image]

It does change the behavior slightly. The panel top corners are now rounded, and the cutout behavior is different (still broken?)
## 2mm
[image]

Changing the cutout to be all 90 degree corners, removing millradius - behavior not fixed.
[image]

### Steps to Reproduce

[kibot-panel.yaml.txt](https://github.com/yaqwsx/KiKit/files/12777281/kibot-panel.yaml.txt) This is the kibot-panel.yaml I am using.

[espairqual.kicad_pcb.txt](https://github.com/yaqwsx/KiKit/files/12777291/espairqual.kicad_pcb.txt) PCB file.

`kibot -c kibot-panel.yaml -b espairqual.kicad_pcb`

yaqwsx, 2023-10-01:
Could you please share the board? I struggle to reproduce this behavior. Just outlines without any traces are sufficient.

yaqwsx, 2023-10-05:
There were two problems:
- the fixed width tab compared floats exactly thus only one part of the tab was generated. Addressed by 3a33bbac4d69bd2b38f65978d79962e2a588bd98
- the partitionline-building algorithm didn't allow for boards with no spacing; this was addressed by 102e4cde39fb78d79efb2dad7b9e910efc7806cd. However, I am only 99% sure about correctness of this solution so it needs further testing before we make a release.

gcormier, 2023-10-05:
Awesome, thanks for looking at this! Apologies about the KiBot file - will convert it to a native file to repro the issue next time.

Can I try out these changes with `nightly` container as described at https://yaqwsx.github.io/KiKit/v1.4/installation/docker/#mac-m1-containers ?

gcormier, 2023-10-05:
Pulled the latest

`pip install git+https://github.com/yaqwsx/KiKit.git`

And looks great!

![image](https://github.com/yaqwsx/KiKit/
[kikit-panel.json.txt](https://github.com/yaqwsx/KiKit/files/12824850/kikit-panel.json.txt)
assets/8388983/e6724988-71f5-4c65-beb6-ec1c59135c5c)

I know you have done it already, but I've attached the kikit json.

set-soft, 2023-10-11:
Hi @gcormier !

Just to let you know that the current KiBot docker images should include @yaqwsx fixes.

The Debian package for KiKit with version 1.4.0-1 contains the above mentioned patches. And the images are created using it. The *kibot-check* tool should report KiKit v1.4.0.1. Note that the *info* output can log the *kibot-check* output, no need to call it manually in a workflow.

gcormier, 2023-10-11:
Oh awesome, I thought I would have to wait until the next major release of KiKit. Great news, thanks! I just ran my GitHub KiBkt pipeline and am getting proper panels created :)

gcormier, 2023-12-01:
I think this is the best place to post this and possibly re-open? You mentioned it might need more testing and this might be related.

I am expecting to see cuts in pink on the 2 right PCB's
[image]

My suspicion is the logic from `cutout` is not checking backbones. It did behave as expected for the top/bottom rails.

Here is the kikit file I'm using.

[kitpanel.json](https://github.com/yaqwsx/KiKit/files/13532287/kitpanel.json)

Here is the PCB - https://raw.githubusercontent.com/gcormier/esphome-pressure/main/pcb/esphome-pressure.kicad_pcb

My command:
`kikit panelize -p kitpanel.json esphome-pressure.kicad_pcb ./out/panel.kicad_pcb`

(This is using latest KiKit installed from pip pointed at git repo.)

yaqwsx, 2023-12-03:
Backbones are not indeed checked (as internally, we don't perform cutout but shape the tab accordingly. We only add material, never remove it). This should be fixed.