yaqwsx/KiKit #598: An error occurred: invalid literal for int() with base 10: '9_4-101'

greenscreenflicker, 2023-10-19:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

kikit, version 1.4.0+6.gd7b2198

### KiCAD version

7.0.7

### Operating system

Windows

### Description

I get the following error

kikit fab jlcpcb --assembly --schematic ../
rp2040_eth.kicad_sch panel.kicad_pcb gerber_panel/ --no-drc
WARNING: Component DPS_Logo_9_4-101 is missing ordercode
WARNING: Component J303 is missing ordercode
An error occurred: invalid literal for int() with base 10: '9_4-101'
No output files produced

### Steps to Reproduce

execute code above with files. Files can be send privately if you need. Tag me when required.

yaqwsx, 2023-10-26:
We assume that references can always be split into character and number parts so we can sort them nicely. You seem to be using reference `DPS_Logo_9_4-101`, which KiKit interprets as `DPS_Logo_` and `9_4-101`. This is why it fails, as it struggles to convert  `9_4-101` into number.

If you need a quick work-around, use sane references.

For a proper solution, I am thinking if just cutting off the last numerical part is sufficient and bullet-proof enough.

yaqwsx, 2023-10-29:
It is bullet proof enough (as it also ensures that the numeric part can be parsed into a number). Fixed in dd6f505.

closed 2023-10-29 by yaqwsx