yaqwsx/KiKit #600: reconstructarcs makes some strange arcs in some layouts

jpeterse, 2023-10-25:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

1.4.0

### KiCAD version

7.0.8

### Operating system

Linux

### Description

Using panelization with tabs, generated by JSON config, with post option reconstructarcs works a mostly to generate some nice output. But sometimes, it fails.
Attached example is a small pcb, consisting only of the edge cut polyline, and the generated output along with the json configuration. Panelized output is generated from CLI.

Example of the issues:

[image]

The top arcs is somehow oddly different from the arc below it. It's the arc below, that I would expect kikit to generate.
 
It equally produces quite odd results where 4 boards meet
 
[image]

### Steps to Reproduce

Using the provided sample files, I run Kikit with:
kikit panelize -p panelize_config.json kikit-panlization-issue-test.kicad_pcb kikit-panlization-issue-test-panelized.kicad_pcb 

[example-of-issue.zip](https://github.com/yaqwsx/KiKit/files/13169211/example-of-issue.zip)

yaqwsx, 2023-10-26:
The implementation of reconstructarcs isn't the best one (as the problem is harder than expected). We already observed this problem in #453.

I can't promise you a fix in the near future; however, there are plans for migrating different geometrical engines that would support arcs out-of-the-box.

jpeterse, 2024-03-07:
that's awesome, thank you for moving this forward.

Let me try and restore my environment back to kicad 7 to check it out
When running my test json against the test pcb with kikit 1.5 and kicad ver. 8 rc1, it just aborts with no further error message. (Running rc1, as my normal install is from the official kicad PPA).

yaqwsx, 2024-03-07:
If it aborts, please, rerun it with debug switch on. If KiCAD or KiKit crashes, a stack trace should be printed.

jpeterse, 2024-03-07:
Ran kikit 1.5 against both kicad 7.11 and 8.0.1-rc1.
Adding debug directives to the json config made no difference. 

> $ /usr/local/bin/kikit panelize -p panelize_config.json kikit-panlization-issue-test.kicad_pcb kikit-panlization-issue-test-panelized-v8.kicad_pcb
> Aborted

Ran pcbnew from console, used the gui to enter the same parameters as in the json file. When running panalize, pcbnew crashes and writes on console:

> $ pcbnew 
> terminate called after throwing an instance of 'std::out_of_range'
>   what():  aGlobalIndex-th vertex does not exist
> Aborted

The error occurs regardless of what reconstructarcs is set to, so I suppose we should open a separate issue on this?

jpeterse, 2024-03-09:
Tested upstream against a number of different panel layouts, and the new arc welder has worked perfectly in all layouts that I have.
I'll open a separate issue for the crash when copperfill has no areas to fill.

yaqwsx, 2024-03-10:
@jpeterse: The warning isn't good. Could you share a board where you experience it?

jpeterse, 2024-03-10:
From memory, it happened on all the boards I tested. It also happens on the board attached to this issue report.

yaqwsx, 2024-03-10:
I struggle to reproduce it on the files you posted.

jpeterse, 2024-03-10:
Odd. I had cleared my sample file, and so I just re-downloaded it from this issue report and ran it again, just before my last comment. Let me try and update to latest upstream kikit and try again.

jpeterse, 2024-03-10:
Updated from g18b590e to g296fc56 and warning is gone.
I'll retest the other projects I have.

jpeterse, 2024-03-10:
Retested my projects, and the warning is now gone.

FYI: I did have one project that failed with:
```
An error occurred: Cannot V-Cut a curve or a line that is either not horizontal or vertical.
Possible cause might be:
- your tabs hit a curved boundary of your PCB,
- your vertical or horizontal PCB edges are not precisely vertical or horizontal.
Modify the design or accept curve approximation via V-cuts.
No output files produced
```

It produced a panel with kikit g18b590e and kicad 8.0.1-rc1, but failed with kikit g296fc56.
All I did to solve this was to open the project in Kicad 8.0.1-rc2 and resave the files. No changes were made in the files.
Just FYI.

yaqwsx, 2024-03-10:
Could you share the project and panelization command (outlines are just fine)? There were indeed changes to the geometry core, and those changes could have unintended consequences for cuts.

jpeterse, 2024-03-10:
I am not able to reproduce the vcut error any more, since I re-saved the project files. I tested the backup files as well.
It's a super simple board outline though.
[image]

I tried to downgrade kikat back to g18b590e, and I can still not reintroduce the error.
Between updates to both kicad and kikit, it's difficult to say what caused the issue.

yaqwsx, 2024-03-15:
I tested the arc welder on a dozen of designs, seems to be working fine.

closed 2024-03-15 by yaqwsx