yaqwsx/KiKit #609: JLC now detects BOM mismatch

gcormier, 2023-11-12:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

latest

### KiCAD version

7.0.8

### Operating system

Github

### Description

As per discussion, JLC will now detect the fiducials/etc as missing

[image]

### Steps to Reproduce

1. Create a panel 
2. Export for fabrication
3. Upload to JLC and proceed to the BOM/CPL step.

[image]

It would be nice to exclude exporting these. I think clicking "continue" is a bad user experience as it might mask an actual human-error during design where there was something not exported properly. It simply trains users to "ignore errors" when using JLC.

[kitritiled-panel_cpl_jlc.csv](https://github.com/yaqwsx/KiKit/files/13328307/kitritiled-panel_cpl_jlc.csv)
[kitritiled_bom_jlc.csv](https://github.com/yaqwsx/KiKit/files/13328308/kitritiled_bom_jlc.csv)
[gerbers.zip](https://github.com/yaqwsx/KiKit/files/13328309/gerbers.zip)

yaqwsx, 2023-11-12:
Could you please provide us with **precise** steps to reproduce (source board and exact command). I struggle to reproduce what you observe. When I run this on any of my projects with the upstream version of KiKit (that is panelization + `kikit fab jlcpcb`), the mousebites, tooling holes and fiducials are not included in the BOM and POS files.

gcormier, 2023-11-12:
Yup, I will put something together for you. I'm using KiBot, so I will make sure to provide a native KiKit config file for you :)

yaqwsx, 2023-11-12:
If you use KiBot, then you are probably not using KiKit export. KiKit has the `fab` command that is tailored for specific fab houses and ensure the output is compatible with them.

gcormier, 2023-11-12:
Hmm good point. The BOM is handled by KiBot. Sorry about that, I will debug a bit further and ask over on the KiBot repo.

closed 2023-11-12 by gcormier

gcormier, 2023-11-15:
https://github.com/INTI-CMNB/KiBot/discussions/516#discussioncomment-7577300

Could you take a look and share your thoughts?