yaqwsx/KiKit #610: Unable to create the panel with vcuts

OndrejBaran, 2023-11-15:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

kikit, version 1.4.0

### KiCAD version

7.0.9

### Operating system

Ubuntu 23.04.1

### Description

I am trying to follow the very basic example for vcut panelize. I a section "Cuts", I entered layer 44 to insert panel shapes to Edge Cuts layer.

I got this Error message:
[image]

If I try any other layer number, I got the same error message with the number mentioned in the error message.

Mousebites work correctly.

### Steps to Reproduce

In the KiKit GUI, fields according to the first example  https://yaqwsx.github.io/KiKit/v1.4/panelization/examples/ were filled.

The layer number in section cuts was set to 44.

kikit panelize \
    --layout 'rows: 2; cols: 2' \
    --tabs 'type: full' \
    --cuts 'type: vcuts; layer: 44' \
    '/home/ondra/Downloads/test.kicad_pcb' panel.kicad_pcb

yaqwsx, 2023-11-15:
Layers should be specified by their name, not their numbers. KiCAD UI only operates on layer names, so there is no reason to support numbers:

[image]

The error message is not helpful and we should improve it.

yaqwsx, 2023-11-15:
This was fixed in b0c1db8. Though, I suggest using layer names instead of numbers as layer numbers are not used in KiCAD UI: 
[image]

closed 2023-11-15 by yaqwsx