yaqwsx/KiKit #611: kikit not found after installation on Linux mint

nopeppermint, 2023-11-15:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

kikit, version 1.4.0

### KiCAD version

Application: KiCad x86_64 on x86_64  Version: 7.0.9-rc2-202311021534~e775a7581a~ubuntu22.04.1, release build  Libraries: 	wxWidgets 3.2.1 	FreeType 2.11.1 	HarfBuzz 6.0.0 	FontConfig 2.13.1 	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.16  Platform: Linux Mint 21.2, 64 bit, Little endian, wxGTK, cinnamon, x11  Build Info: 	Date: Nov  2 2023 15:34:22 	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24 	Boost: 1.74.0 	OCC: 7.5.2 	Curl: 7.88.1 	ngspice: 38 	Compiler: GCC 11.4.0 with C++ ABI 1016  Build settings: 	KICAD_SPICE=ON

### Operating system

Linux Mint

### Description

After installation of kikit, kikit is not usable in the command line on linux, additional steps are needed to get it working in the command line

### Steps to Reproduce

Install Kicad 7.0.9 from the ubuntu ppa "sudo add-apt-repository ppa:kicad/kicad-7.0-releases" on Linux Mint 21.2

Install Kikit via PCM

Install Kikit backend via "pip3 install kikit"

try to run ```kikit``` in the terminal

I always got the error:

```
Der Befehl 'kikit' wurde nicht gefunden, meinten Sie:
  Befehl 'kinit' aus dem deb krb5-user (1.19.2-2ubuntu0.2)
  Befehl 'kinit' aus dem deb heimdal-clients (7.7.0+dfsg-3ubuntu1)
Versuche: sudo apt install <deb name>

```
I found others reporting the same issue https://forum.kicad.info/t/solved-help-with-installing-kikit-on-linux/34092/2

The solution is:

https://stackoverflow.com/questions/35898734/pip-installs-packages-successfully-but-executables-not-found-from-command-line/59436732#59436732

more precise:

in case of python 3.11:

```python3.11 -m site &> /dev/null && PATH="$PATH:`python3.11 -m site --user-base`/bin"  ```

in case of another python 3 version, adapt the version in the string

yaqwsx, 2023-11-15:
This doesn't seem like KiKit issue; it is just that your OS packages don't add Python packages into `PATH`.

nopeppermint, 2023-11-15:
fair enough

closed 2023-11-15 by nopeppermint