yaqwsx/KiKit #614: Changing resistor designators manually leads to non-assembled components

greenscreenflicker, 2023-11-23:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0+26.g56c696a

### KiCAD version

7.0.7

### Operating system

Windows 11

### Description

I have  a multi project PCB. I want to swap out some resistors to have another output voltage configuration. However, when running the panalizing command, the components are missing.
@yaqwsx I send you all the files via email.

### Steps to Reproduce

see above

greenscreenflicker, 2023-11-23:
1. Go to folder panel in KiCad Command Prompt

2. Execute the command "kikit panelize  --layout "grid; rows: 3; cols: 3; space: 0mm; rotation: 90deg;alternation: cols;"  --tabs "fixed; width: 20mm; vcount: 1; hcount: 0"  --cuts "vcuts"   --post "millradius: 1mm"  --source "tolerance: 5mm" --text "simple; text: Digital Power Systems, Karlsruhe, DCC48XX, rc5 production run; anchor: mt; voffset: 2mm; hjustify: center; vjustify: center;"  --framing "railstb; width: 4mm; space: 2mm;" --fiducials "3fid; hoffset: 2mm; voffset: 2mm; coppersize: 2mm; opening: 1mm;" ../DCC6024UT.kicad_pcb panel.kicad_pcb"

3. Go to the gerber panel folder, search for R1041. You won't see something. Problem: But it should be there.

4. Open the Panel in PCB new, search for R1041. You will see the resistor.

5. Open the schematic in the base folder, search for R1041. You will see the resistor.

yaqwsx, 2024-01-06:
I believe this was already resolved in the original e-mail conversation.

closed 2024-01-06 by yaqwsx