yaqwsx/KiKit #615: Vcuts should be placed at top positions

greenscreenflicker, 2023-11-24:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

kikit, version 1.3.0+16.gf84d28c

### KiCAD version

7.0.7

### Operating system

Windows 11

### Description

The Vcuts are placed in the middle. Is there an option to place the vcuts at the top of a pcb, without specifying it manually?

### Steps to Reproduce

Run the panalizing command with the pcb you already have from me.

[image]

Observe the vcuts, that are placed in the middle not, in the upper most position. Is there an option to specify this?

yaqwsx, 2023-11-24:
Any cuts are placed only on places where the tabs are. In your case, you have a tab in the middle of the board. And the V-cut is there to cut the tab-board interface. You should position your tabs correctly (e.g., via annotation).

greenscreenflicker, 2023-11-24:
Exactly, i could specify it manually. (see top: Is there an option to place the vcuts at the top of a pcb, without specifying it manually?) 
But could there be an option to automatically detect, that it's a vcut and always use the topest/lowest edge here? (it will be always true for vcuts, hence my proposal)

yaqwsx, 2023-11-24:
KiKit is tab-oriented. Any cuts produced are generated from tabs. What you probably want is a new algorithm for tab placement: something like "edge tab" that will find the "outer-most" edge and placed tab there (which is a good idea and we should have it). It just shouldn't be coupled with V-cuts.

greenscreenflicker, 2023-11-24:
@yaqwsx good proposal to keep it more versatile! +1 from my side.

yaqwsx, 2024-01-06:
When revisiting this, I just realized that this is exactly what the full tab does. It finds the outer most edges and spans a tab from them. So edges tabs are redundant.

closed 2024-01-06 by yaqwsx