yaqwsx/KiKit #618: KiKit crashes on PCB with castelations

bem22, 2023-12-02:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

1.4.0

### KiCAD version

7

### Operating system

Linux - UBUNTU 23.04

### Description

I followed the guide and when I click Panelize, Pcbnew crashes

### Steps to Reproduce

Using my kicad_pcb and multi.json click panelize

[multi.json](https://github.com/yaqwsx/KiKit/files/13537339/multi.json)
[wembed-systems-bms-v2.zip](https://github.com/yaqwsx/KiKit/files/13537341/wembed-systems-bms-v2.zip)

yaqwsx, 2023-12-03:
I just tested your panel on `kikit, version 1.4.0+20.gbb263ac` (current upstream) and it works with no problems:

[image]

Please, use the upstream version or wait for the next release.

closed 2023-12-03 by yaqwsx

bem22, 2023-12-16:
It still crashes for me with the latest commit from the main branch

yaqwsx, 2023-12-16:
Any chance that you use Shapely 2.0?

bem22, 2023-12-16:
[image]