yaqwsx/KiKit #619: Add the option to relabel the component names such that they are unique

bem22, 2023-12-03:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

When using OpenPnP, it would be great if the .pos file generated by KiCad after panelization with KiKit had unique ids. 

The new boards are "generated" with KiKit so I think this is the point where the parts could receive new names?

yaqwsx, 2023-12-03:
Could you provide (even a simple) example? What are "IDs"? References? Making the references unique across panels is not a good idea, as it will also change the silkscreen.

Generally, I am familiar with two workflows (on industrial machines):
- you have fiducials on each PCB in the panel. Then you usually need only POS file for single board and you place it multiple times in the PnP
- you have fiducials only on the panel frame. Then you need a single POS file for the whole panel. However, having duplicate references isn't a problem as the corresponding components are the same.

Please, describe your use case more in detail. Then we can design the feature appropriately.

bem22, 2023-12-03:
[image]
I would assume this is the part of the file that is interesting for parsing.

This gives the position, the Reference and the package.

What would be the workflow to integrate this into KiKit? And I assume this is python?

yaqwsx, 2023-12-03:
You missed my point - we shouldn't change how we build the panel. We should provide specialized fabrication output for OpenPNP.

bem22, 2023-12-03:
I got that. I meant what would the workflow be?

Generate a .pos file from the bottom up? Generate a file using the KiCad generator, parse it and replace the duplicates with sensible names like `board-{n}-{originalRef}`

Generate a different output file than the one made by KiCad?

That's what I'm asking

bem22, 2023-12-16:
One question, what does "renameRefs" function do exactly?

yaqwsx, 2023-12-19:
> renamenet, renameref: A pattern by which to rename the nets and references. You can use {n} and {orig} to get the board number and original name. Default values are Board_{n}-{orig} for nets and {orig} for references.
(https://yaqwsx.github.io/KiKit/v1.4/panelization/cli/#layout)

bem22, 2023-12-21:
This should... in principle solve my request?

I'll check it out today.

yaqwsx, 2024-01-06:
OpenPNP fabrication output was added in db18615

closed 2024-01-06 by yaqwsx