yaqwsx/KiKit #622: Allow V-Cuts to only horizontal or vertical

bradanlane, 2023-12-07:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

For completely odd shaped PCBs, mousebites are the easy solution.

There are cases where odd shaped PCBs may have opposing sides which are either horizontal or vertical. (see image below).

Suggestion: allow V-Cuts to be applied to vertical xor horizontal.

I tried a few different methods. The closest I was able to get was to use framing = tightframe and tabs with hcount = 1 and vcount = 0. I was not able to find any solution to only have vertical v-cuts v-cuts on the ends.

I use the JSON interface.

[image]

Here is the JSON I have thus far ...
```
{
    "layout": {
        "type": "grid",
        "rows": 1,
        "cols": 1,
        "hspace": "2mm",
        "vspace": "2mm"
    },
    "source": {
        "type": "auto",
        "tolerance": "2mm"
    },
    "tabs": {
        "type": "fixed",
		"hwidth": "61mm",
		"hcount": "1",
		"vcount": "0"
    },
    "framing": {
        "type": "tightframe",
		"hspace": "5mm",
		"vspace": "5mm",
		"slotwidth": "2.5mm",
        "width": "2.5mm"
    },
	"tooling": {
		"type": "4hole",
		"hoffset": "3.5mm",
		"voffset": "3.5mm",
		"size": "1.8mm"
	},
	"fiducials": {
		"type": "4fid",
		"paste": true,
		"hoffset": "13.5mm",
		"voffset": "3.5mm",
		"coppersize": "1mm",
		"opening": "1.5mm"
	},
	"post": {
		"type": "auto",
		"millradius": "1mm",
		"origin": "bl",
        "copperFill": false
	}
}
```

yaqwsx, 2024-02-21:
The intended workflow for KiKit is:
- you specify, where tabs should be placed (via manual or automatic annotation),
- the annotations are turned into tabs.
- The tabs are the the only thing connecting boards to the rest of the panel and from the tabs. Required cuts are automatically deduced from the tabs.

Therefore, what we miss to fulfull your request is to have an automatic annotation placement of the full-width tabs that only uses left/right or top/bottom sides.

At the moment a possible work-around is to use fixed-width tabs (with width of the wider side).

bradanlane, 2024-02-21:
In my OP I did mention trying fixed width tabs. It creates the tabs but I still must manually add the v-cuts.

I am also still suffering the `'SwigPyObject' object has no attribute 'FootprintLoad'` error (issue #571)

I tested using kikit 1.4.0+23.gdb18615 and 1.4.0+44.gba5a426

yaqwsx, 2025-09-16:
@DunningtonAudio Could you share the configuration you used? The second image should be doable if you do not place any tabs between the boards. And it should be a fairly common use case for KiKit.

DunningtonAudio, 2025-09-19:
@yaqwsx Thanks, the exported json is:

`{
    "layout": {
        "hspace": "2mm",
        "vspace": "2mm",
        "hbackbone": "16mm",
        "rotation": "90deg",
        "rows": "2",
        "cols": "5"
    },
    "tabs": {
        "type": "fixed",
        "vwidth": "5mm",
        "hwidth": "1mm",
        "hcount": "0"
    },
    "cuts": {
        "type": "vcuts"
    },
    "framing": {
        "type": "railstb",
        "width": "8mm"
    },
    "tooling": {
        "type": "4hole",
        "hoffset": "2mm",
        "voffset": "2mm"
    },
    "fiducials": {
        "type": "3fid",
        "hoffset": "5mm",
        "voffset": "2mm"
    }
}`

Everything else is default values. The only thing that seemed odd to me is that a value of 0 is not accepted for /tabs/hwidth, although hcount is set to 0 and as you can see, there are no horizontal tabs, so I'm not sure what setting should be changed to achieve what you're suggesting?