yaqwsx/KiKit #625: pip install kikit fails on Mac

zapta, 2023-12-27:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

1.4.0

### KiCAD version

7.0.1-0

### Operating system

Mac 13.14.1  Apple M2

### Description

Pip installation fails.

```

/Users/user $ python --version
Python 3.12.1
/Users/user $ pip --version
pip 23.3.2 from /Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/pip (python 3.12)
/Users/user $ pip install kikit
Collecting kikit
  Using cached KiKit-1.4.0-py3-none-any.whl.metadata (5.7 kB)
Collecting click>=7.1 (from kikit)
  Using cached click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
Collecting commentjson>=0.9 (from kikit)
  Using cached commentjson-0.9.0.tar.gz (8.7 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting markdown2>=2.4 (from kikit)
  Using cached markdown2-2.4.12-py2.py3-none-any.whl.metadata (2.0 kB)
Requirement already satisfied: numpy in /Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages (from kikit) (1.26.2)
Collecting pcbnewTransition<=0.4,>=0.3.4 (from kikit)
  Using cached pcbnewTransition-0.3.4-py3-none-any.whl (6.7 kB)
Collecting pybars3>=0.9 (from kikit)
  Using cached pybars3-0.9.7.tar.gz (29 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting shapely<2.0,>=1.7 (from kikit)
  Using cached Shapely-1.8.5.post1.tar.gz (200 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [33 lines of output]
      Traceback (most recent call last):
        File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 112, in get_requires_for_build_wheel
          backend = _build_backend()
                    ^^^^^^^^^^^^^^^^
        File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 77, in _build_backend
          obj = import_module(mod_path)
                ^^^^^^^^^^^^^^^^^^^^^^^
        File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/importlib/__init__.py", line 90, in import_module
          return _bootstrap._gcd_import(name[level:], package, level)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
        File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
        File "<frozen importlib._bootstrap>", line 1310, in _find_and_load_unlocked
        File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
        File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
        File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
        File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
        File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
        File "<frozen importlib._bootstrap_external>", line 994, in exec_module
        File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
        File "/private/var/folders/6v/ck129h0s0g77x56llnrrfgwc0000gn/T/pip-build-env-vpl9md8r/overlay/lib/python3.12/site-packages/setuptools/__init__.py", line 16, in <module>
          import setuptools.version
        File "/private/var/folders/6v/ck129h0s0g77x56llnrrfgwc0000gn/T/pip-build-env-vpl9md8r/overlay/lib/python3.12/site-packages/setuptools/version.py", line 1, in <module>
          import pkg_resources
        File "/private/var/folders/6v/ck129h0s0g77x56llnrrfgwc0000gn/T/pip-build-env-vpl9md8r/overlay/lib/python3.12/site-packages/pkg_resources/__init__.py", line 2191, in <module>
          register_finder(pkgutil.ImpImporter, find_on_path)
                          ^^^^^^^^^^^^^^^^^^^
      AttributeError: module 'pkgutil' has no attribute 'ImpImporter'. Did you mean: 'zipimporter'?
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
/Users/user $ 
```

### Steps to Reproduce

run the command below on a similar system.

pip install kikit

yaqwsx, 2023-12-27:
KiKit relies on Shapely <= 2.0; however, such shapely is incompatible with Python 3.12. This is the reason for the error message. At the moment, there is no work-around; we probably have to introduce support for Shapely 2.0, but it can take a while.

zapta, 2023-12-27:
Thanks for the quick answer. What is the latest Python version that is supported?

Also, I suggest to add a note in the installation instructions. https://github.com/yaqwsx/KiKit

yaqwsx, 2024-01-06:
We started supporting Shapely 2. If you encounter any problems with Shapely 2, please let us know in #574.

closed 2024-01-06 by yaqwsx