yaqwsx/KiKit #626: VCUT support on panelizing boards with indents.

greenscreenflicker, 2024-01-05:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

I have a board with indents/cutouts at the side. KiKit does not seem to want to panallize the idented site (marked with arrows) with full side vcuts.
Is there a possibility to add support on this?
[image]

yaqwsx, 2024-01-05:
Can you provide us with a config you tried? It should work, it even adds cutouts: #346

greenscreenflicker, 2024-01-05:
@yaqwsx thanks for the promt respone.
`kikit panelize ^
    --layout "grid; rows: 2; cols: 2; space: 2mm; rotation: 90deg;" ^
    --tabs "fixed; width: 10mm; vcount: 2; hcount: 5;" ^
    --cuts vcuts ^
    --post "millradius: 1mm" ^
    ../mypcb.kicad_pcb panel.kicad_pcb`

yaqwsx, 2024-01-05:
I guess that you place a tab such that it hits the curved surface. Which yields a curved cut that cannot be V-scored (the error message should say that). If you want V-cuts, I suggest using full-width tabs. Or enable the option to V-cut curves (cutcurves: true).

greenscreenflicker, 2024-01-05:
Fair enough, full width tabs. Did that. Same issue.

> kikit panelize ^
    --layout "grid; rows: 2; cols: 2; space: 2mm; rotation: 90deg;" ^
    --tabs "full" ^
    --cuts vcuts ^
    --post "millradius: 1mm" ^
    ../mypcb.kicad_pcb panel.kicad_pcb
An error occurred: Cannot perform V-Cut which is not horizontal or vertical
No output files produced

yaqwsx, 2024-01-05:
Without sharing the design I can only guess. Maybe your edges are not perfectly vertical or horizontal? Nevertheless, if you add the option to cut curves, you will see the output.

greenscreenflicker, 2024-01-05:
@yaqwsx I send you the files. Thanks already for your prompt support.

yaqwsx, 2024-01-06:
I don't want to be rude; I already gave you the answer - your edges are not perfectly vertical or horizontal. Couldn't you check that before sending me the files (also, the steps for reproduction were flawed as there were two identical commands).

[image]

greenscreenflicker, 2024-01-06:
Thank you for pointing that out. You are absolutely correct.
Let me propose an improvement to not frustrate users: 
A warning message: "Attention, board lines (starting: xy, stopping xy) are not on same Y axes. However this is required for the vcut feature".
Maybe you could incoperate this kindely?

yaqwsx, 2024-02-21:
I don't we can do better than "Cannot perform V-cut which is not horizontal or vertical" as it is quite hard to algorithmically detect your intention and validate what is wrong. The root cause is that you arranged your panel such that it would yield a non-orthogonal V-cut.

I added some hints about what might lead to non-orthogonal V-cut in 970db7af2ae5417150436f7d2076aaa685fb8904.

greenscreenflicker, 2024-02-21:
Sorry to bother again, could you add the coordinates in the error message please?
That would help the user.

yaqwsx, 2024-02-21:
I can only give you coordinates in the panel, which I am not sure is useful at all (as the panel generator doesn't finish). There is no relation between cuts and the original substrate (as there might be no relation at all).

So if panel coordinates help, sure, we can add them.

greenscreenflicker, 2024-02-21:
As panel won't be outputted, i guess it won't help.
How about a nearby component? E.g. "vcut close to 'R12'" or so?

yaqwsx, 2024-02-21:
The recommended aid is to re-run the command with rendering cuts to a layer. Then, true, coordinates in panel space might help.

greenscreenflicker, 2024-02-21:
I mean panel cordinates are better than nothing, but components designators, like R12, would be more usefull.

yaqwsx, 2024-02-21:
I am afraid that it is not generally applicable. What if you panel a PCB with no or a few components that are far-far from the actual problem? That's only misleading. We need solutions that are generally applicable, not small band-aids for problems you encounter.

I have another solution in mind, but that requires quite a lot of implementation effort and it won't happen soon: #640

greenscreenflicker, 2024-02-21:
I like that approach. It's better than my proposal. And should be defentively favored.
How about just making a PCB with the extention "_failed" (to not to be accenidently used by users like me) and placing a silkscreen marker there.