yaqwsx/KiKit #635: NETINFO_LIST object has no attribute RemoveUnusedNets

TimGoll, 2024-02-15:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

GitHub master @6dc6666

### KiCAD version

KiCad 8.0 RC2

### Operating system

Windows 11

### Description

[image]

When trying to panalize a board I get this error. The same board worked in KiCad 7.0.10. So I'm pretty sure that is related to python binding name changes

### Steps to Reproduce

Panelize a board, the boards I tried should not be published, therefore I can not attach them here.

TimGoll, 2024-02-15:
Same thing happens on the most recent build: `kicad-nightly-8.0.0.rc2.569.g2c9c741895-x86_64`

yaqwsx, 2024-02-15:
Fixed in a49b818.

closed 2024-02-15 by yaqwsx

TimGoll, 2024-02-15:
Awesome, thank you so much!

TimGoll, 2024-02-15:
I'm adding this here as this is probably related: 
[image]

yaqwsx, 2024-02-15:
Ah, I used inapropriate testing board. Fixed in e0a347a.

TimGoll, 2024-02-15:
It works now, but still throws an error:
[image]

TimGoll, 2024-02-19:
I just wanted to comment again to see if you've seen my last comment. As I think this is still releated to the initial issue, I did not want to open a new one but I suspect my comment got lost in your notifications. Thanks!

yaqwsx, 2024-02-19:
I've seen it, but I don't have a quick fix at the moment. It is only related to GUI, it has nothing to do with the panelization feature per se. Howerver, it seems that the essential API is gone in v8. This will probably require fix on KiCAD's side.

yaqwsx, 2024-02-24:
It seems that the API is no longer needed as KiCAD started to do some of the housekeeping automatically. Reflected in 611dc10