yaqwsx/KiKit #637: foorprint with arc Edge Cut lines create problems

JulienCarayon, 2024-02-19:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

1.4.0

### KiCAD version

v7.0.10

### Operating system

macOs Ventura 13.0.1

### Description

Kicad 7 issue :

swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'FP_SHAPE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
An error occurred: unsupported operand type(s) for +: 'SwigPyObject' and 'VECTOR2I'
No output files produced
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'std::vector< ZONE * > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.

Here is a problematic footprint exemple :
[image]

[problematic.txt](https://github.com/yaqwsx/KiKit/files/14334746/problematic.txt)
(rename the .txt to .kicad_mod)

### Steps to Reproduce

Use the footprint linked above and try to split board with a board containing the footprint

yaqwsx, 2024-03-15:
This might be related to #663 - both reports are form mac, and they contain numerous errors in KiCAD API.

yaqwsx, 2025-01-26:
Could we retry this with the upcoming KiCAD 9?