yaqwsx/KiKit #638: Clarify what instance of python/pip are used for KiKit

ostenning, 2024-02-20:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

It is a little unclear from the current README to which python installation is being used. It also may be unclear to the user who isn't familiar with KiCad plugins that KiCad has its own python instance.

I recommend adding some clarifications rather than having people simply run `pip3 install kikit` as the README states.

Best regards

yaqwsx, 2024-02-20:
We have quite step-by-step instructions for installation: https://yaqwsx.github.io/KiKit/v1.4/installation/intro/ Those instructions are even linked from README. 

Could you be more specific which part of the installation guide should be improved?

ostenning, 2024-02-21:
My feedback is only for the README:

> Installation
> 
> KiKit is available as a [PyPi package](https://pypi.org/project/KiKit/), therefore, you can install it via pip:
> 
> `pip3 install kikit`
> 
> Then you can install the Pcbnew action plugins via KiCAD's PCM.

`pip3 install kikit` does not specify the KiCad pip instance, so running this command will install KiKit for the default OS python instance and then running it results in errors as there are no references to KiCad binaries (like Pcbnew etc).