yaqwsx/KiKit #639: Pannelization fails on Kicad 8

xobs, 2024-02-20:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

1.4.0

### KiCAD version

8.0.0-rc3

### Operating system

Windows

### Description

When clicking "Panelize", the following error is displayed:

[Window Title]
Error

[Main Instruction]
Cannot perform:

[Content]
PCB_TEXT.GetShownText() missing 1 required positional argument: 'aAllowExtraText'

[OK]

[image]

### Steps to Reproduce

1. Install KiCad 8
2. Install kikit from the plugins manager
3. Install kikit cli from pip when instructed
4. Attempt to panelize a PCB

yaqwsx, 2024-02-20:
This is a duplicate of #636 - v1.4 **doesn't support v8**, you have to use the upstream version. We are still in the phase of bug hunting, stable release is planned to be synchronized with the KiCAD 8 release.

closed 2024-02-20 by yaqwsx

xobs, 2024-02-21:
I opened a new issue since #636 was closed, and the issue still occurs.

yaqwsx, 2024-02-21:
You reported a problem in version v1.4. This version doesn't support v8 and never will.

There is an upstream version with experimental v8 support. This will be released as v 1.5 in a few weeks. If there is a problem in the upstream version, please, file a bug report for the upstream version.

PS: #636 was closed after you reported this issue.