yaqwsx/KiKit #641: JLCPCB: Avoid misleading error messages by optimized bom

greenscreenflicker, 2024-02-21:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

kikit, version 1.4.0+20.gbb263ac

### KiCAD version

7.0.7

### Operating system

Windows

### Description

I get the following error message when ordering at JLCPCB when i use KiCads PCBNews DNP feature.
Could you kindly optimize the BOM, so that error messages doesn't appear?

[image]

### Steps to Reproduce

Make a project, mark one component as DNP, upload to jlcpcb, oberve error msg.

yaqwsx, 2024-02-21:
This bug was fixed in bb87462.

closed 2024-02-21 by yaqwsx

greenscreenflicker, 2024-02-21:
Thank you for your fast reaction.