yaqwsx/KiKit #644: Separate: Edge.Cuts line width is not preserved

Arxi, 2024-02-24:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

1.4.0+46.g5445c99

### KiCAD version

8.0.0

### Operating system

MacOS Sonoma 14.1.2

### Description

After a board is separated from multiboard design using `kikit separate`, the line widths of lines and arcs (and probably other shapes) on Edge.Cuts layer are reset from their original values in multiboard file to 0.1mm. Different PCB manufacturers specify different line widths for board outline and cutouts, for example [PCBWay](https://www.pcbway.com/blog/Engineering_Technical/Milling_of_PCB.html): 
> The inner and outer contours of the circuit board should be indicated with a line width of 1µm! This prevents problems in dimensioning and measurement.

or [JLCPCB](https://jlcpcb.com/help/article/14-Instructions-for-ordering):
> Usually, we suggest our clients design a thickness of 0.15mm for the board outline, slots, cutouts, and milling

Line width of shapes on Edge.Cuts should thus be preserved, as it might be important for certain PCB manufacturers.

### Steps to Reproduce

1. unzip and open [cutouts.kicad_pcb.zip](https://github.com/yaqwsx/KiKit/files/14393938/cutouts.kicad_pcb.zip)
2. see the line widths of rectangular cutouts on the left side of TOP PCB: some are 0.001mm, some 0.005mm, some 0.05mm and some 0.2mm (ignore the rest of the file)
4. use `kikit separate --looseArcs --source 'annotation; ref: TOP' cutouts.kicad_pcb TOP.kicad_pcb`
5. open TOP.kicad_pcb and see all the line widths are 0.1mm

Arxi, 2024-02-24:
Also, I'm not sure if this is relevant, but I noticed that when I run DRC in the resulting PCB `TOP.kicad_pcb`, I get many Edge.Cuts errors which weren't present in the original multiboard `cutouts.kicad_pcb`.

[image]

Seems like the original arcs are replaced with small polygon segments, some of which have 0nm length, and that causes the errors.

Is this because of the `--looseArcs` parameter? I can't drop the parameter as the cutouts then get weird shapes (which is because of [issue 453](https://github.com/yaqwsx/KiKit/issues/453) I guess)

yaqwsx, 2024-02-24:
To give you context:

KiKit internally works by reconstructing 2D polygons from the edge cuts. The edge cuts are lost, we perform all penalization on polygons. This approach allows us to construct tabs, simulate milling, and other operations. Just before the panel is saved we render the polygon back into Edge.cuts.

We use Shapely on the background to handle polygons. This is the reason why we lose arcs (there are plans to switch to a different geometry engine that could handle arcs). The option `reconstructArcs` tries to restore the arcs in the polygons, but it doesn't work well, so it is not recommended for usage. 

The separate command is simply implemented as "create a panel 1x1 with no extra features" as we don't have to reimplement any selection logic.

Regarding this issue:
- I plan to resolve the problem of losing arcs by switching to a different geometry engine in the future;
	- though the DRC error seems to be new in KiCAD 8. So far KiCAD has been happy with short-line segments.
- There are no plans for preserving the original width of the outline. There is no relation between the outline in the source board and in the panel (in principle there is no 1:1 mapping, we even create new outlines!)
	- we could add an option to specify the line width in the final panel, and
	- if needed we could distinguish internal and external perimeters (but I struggle to find a good reason why anyone should want this).
- regarding PcbWays - this is where I get opinionated. Their suggestion to use 1µm lines is just silly IMHO and I am not willing to respect that. There is only a single way to reasonably interpret a cut line - like a hairline line, not its graphical representation. Its thickness is just a "nice way" to represent it reasonably for humans in CAD programs. I understand their paranoia as I've seen electrical engineers draw slots like single thick line, however, I find it pretty messy and unsystematic (as you mix the semantics of lines in a single layer).

Arxi, 2024-02-24:
Thank you for all the context and explanations! I see.

I think the ability to specify unified line width in the final separated file would be enough. I agree that having multiple line widths doesn't seem to be necessary.

As for the DRC error - it seems there's always only one such error for a polygon / shape, which suggests Kicad is fine with most of the short-line segments, but there's one "0nm" segment for each shape which fails the DRC. At least that's what Kicad says. 

Curiously enough, when I tried the `kikit fab jlcpcb --no-drc TOP.kicad_pcb production/` the shapes of the cutouts in the produced gerber files seem fine, when I look at them in Kicad's gerbview. But when I upload them to JLCPCB, their system fails to recognize the board edges (I see nothing in their preview and the board dimensions are auto-calculated incorrectly).

[image]

yaqwsx, 2024-03-06:
As the switch to a new geometry engine will take some time, I tried to fix the current arc welder in a2c9891. The upstream version should fix your issue.

Arxi, 2024-03-09:
@yaqwsx I'm trying this out, but I'm not sure which issue you mean - I see that now I can drop the `--looseArcs` parameter and the polygons are reconstructed correctly (which is nice, thank you!), but the line width is still re-set to the default 0.1mm. 

Nevertheless, after this discussion, I don't see the line width as an important issue.

yaqwsx, 2024-03-17:
The option for specifying the `Edge.Cuts` width was added in a0c1e44.

closed 2024-03-17 by yaqwsx