yaqwsx/KiKit #647: kikit kicad8 installation error

greenscreenflicker, 2024-02-27:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

current/master

### KiCAD version

8.0.0

### Operating system

windows

### Description

when trying to install, kikit cannot be found from kicad console.

### Steps to Reproduce

1. Install KiCad 8.0.0 using the windows installer
2. Open kicad terminal
3. install Kikit using pip install git+https://github.com/yaqwsx/KiKit@master
4. Try to execute "kikit --version" -> KiKit cannot be found

yaqwsx, 2024-02-27:
Try running `python -mkikit.ui --version` instead. It is possible that KiCAD 8 on Windows doesn't include Python entry points in PATH:

greenscreenflicker, 2024-02-27:
works for now. Thanks. How to fix this for average joe?

yaqwsx, 2024-02-27:
You have to modify the PATH variable or report bug in Windows packaging of KiCAD.

greenscreenflicker, 2024-02-27:
Bug reported to KiCAD
https://gitlab.com/kicad/code/kicad/-/issues/17156

yaqwsx, 2024-03-02:
Based on comments in KiCAD repository this is resolved and it will be a part of the next patch release. Closing.

closed 2024-03-02 by yaqwsx

SimonMerrett, 2024-05-17:
@yaqwsx I've just used `pip install kikit` on KiCad 8.0.2 (upgraded from 8.0.0 this morning). I did not have kikit on 8.0.0. I get the warning
```
 WARNING: The script markdown2.exe is installed in 'C:\Users\name\Documents\KiCad\8.0\3rdparty\Python311\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The scripts kikit-info.exe and kikit.exe are installed in 'C:\Users\name\Documents\KiCad\8.0\3rdparty\Python311\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
```
and `kikit --help` gets the response `'kikit' is not recognized as an internal or external command, operable program or batch file.`
So I don't think it's fixed in the 8.0.2 patch on Win10.