yaqwsx/KiKit #652: Configurable V-Cut lines overshoot length and thickness (JLCPCB dimension calculation bug)

markh-de, 2024-03-01:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

It would be very nice if KiKit would allow **configuring the V-Cut line overshoot** (i.e. extra) length and the **V-Cut line thickness**.

At the moment these values are hard-coded to 3 mm overshoot and 0.4 mm thickness.

Reason:

I don't know since when, but currently JLCPCB takes V-Cut lines and even their thickness into account when calculating the board size, which is used for quoting the price of your order.

*Note:* Even if you don't use V-Cuts, the thickness of normal Edge.Cuts lines is also taken into account. For example, if you upload a simple board of 70 × 70 mm² with 0.15 mm Edge.Cuts lines, it will be calculated as 70.15 × 70.15 mm². This is not going to cost you a lot more, but in the case of overshooting V-Cut lines, the cost increase can be quite significant, as you pay for virtual extra 6.4 mm per dimension (X/Y).

For example, being able to reduce the V-Cut line thickness to 0.05 mm (as opposed to the 0.1 mm geometry outline) and the V-Cut overshoots to 0.5mm per side (incl. the radius at the line end) would reduce the calculated extra size (and cost) to 1mm per dimension, which would be acceptable, imho.

Of course, an overshoot of zero would be a bad idea, because then the V-Cut lines could be confused with the real substrate geometry.

JLCPCB is aware of the problem and recommends to note the correct board size in the order comments and request a refund for overpayed boards.

yaqwsx, 2024-03-17:
To be honest, I don't see how implementing any of this could help you to work-around JLC PCB bugs. Also, I think we should patch KiKit in crazy ways just to work around JLC PCB bugs. Instead, we should push them to fix them.

Nevertheless, I was about to implement #601, so I added options to fine tweak the design of V-Cuts in 2e25a3d.

PS: There is `--post `dimensions: true` to render size of the board into the panel.

closed 2024-03-17 by yaqwsx

markh-de, 2024-03-18:
Hi Jan,

I've had two chat sessions with the JLC support about their board size calculation taking irrelevant objects or line widths into account. I kindly asked them to create a ticket for this issue, but of course I don't know if they really care.

I don't think allowing users to tweak the V-Cut style is a crazy patch (or dirty hack). This is a very nice feature allowing people to customize their panel layout. So thank you very much for implementing it upstream! As soon as this is released, I can remove my custom post-processing code (reducing V-Cut lines by 3mm on all ends I'd call a much crazier way of fixing things :wink:).

Your code indeed helps working around the issue that I described. By reducing the V-Cut lines, their board size calculation's error gets into the neglectable range.