yaqwsx/KiKit #654: unhashable type: 'BOARD_ITEM'

greenscreenflicker, 2024-03-06:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I use KiKit at least version 1.4.0 (older version are not supported)

### KiKit version

python -m kikit.ui, version 1.5.0

### KiCAD version

8.0.0

### Operating system

windows

### Description

The code returns an error message. An error occurred: unhashable type: 'BOARD_ITEM'

python -mkikit.ui fab jlcpcb --assembly --schematic ../dcdcpower.kicad_sch ../dcdcpower.kicad_pcb single

Any recommendation how to fix?

### Steps to Reproduce

Execute code aboth.

yaqwsx, 2024-03-06:
Please, provide the files.

greenscreenflicker, 2024-03-08:
@yaqwsx could you please check? I would like to order the PCBs.

yaqwsx, 2024-03-10:
I just tested this with KiKit version 1.5.0+12.g296fc56 and KiCAD 8.0.1-rc2 with the following output:
```
** Found 7 DRC violations: **
[solder_mask_bridge]: Front solder mask aperture bridges items with different nets
    Rule: board setup solder mask min width; Severity: error
    @(181.1500 mm, 92.4500 mm): Polygon on F.Cu
    @(180.8250 mm, 92.3500 mm): Pad 1 [/Uin] of C302 on F.Cu
[solder_mask_bridge]: Front solder mask aperture bridges items with different nets
    Rule: board setup solder mask min width; Severity: error
    @(181.1500 mm, 91.7000 mm): Polygon on F.Cu
    @(180.8250 mm, 92.3500 mm): Pad 1 [/Uin] of C302 on F.Cu
[footprint_type_mismatch]: Footprint component type doesn't match footprint pads (expected 'Through hole'; actual 'SMD')
    Local override; Severity: error
    @(175.2000 mm, 43.7000 mm): Footprint U601
[clearance]: Clearance violation (netclass 'Default' clearance 0.1300 mm; actual 0.0556 mm)
    Rule: netclass 'Default'; Severity: error
    @(180.3100 mm, 34.2600 mm): Polygon on F.Cu
    @(175.0600 mm, 31.8500 mm): Zone [GNDPWR] on F.Cu
[clearance]: Clearance violation (netclass 'Default' clearance 0.1300 mm; actual 0.0000 mm)
    Rule: netclass 'Default'; Severity: error
    @(181.1500 mm, 92.4500 mm): Polygon on F.Cu
    @(167.2423 mm, 76.0629 mm): Zone [/Uin] on F.Cu
[clearance]: Clearance violation (netclass 'Default' clearance 0.1300 mm; actual 0.0000 mm)
    Rule: netclass 'Default'; Severity: error
    @(186.1000 mm, 95.3500 mm): Polygon on F.Cu
    @(158.1300 mm, 26.2400 mm): Zone [GNDPWR] on F.Cu and B.Cu
[clearance]: Clearance violation (netclass 'Default' clearance 0.1300 mm; actual 0.0000 mm)
    Rule: netclass 'Default'; Severity: error
    @(181.1500 mm, 91.7000 mm): Polygon on F.Cu
    @(167.2423 mm, 76.0629 mm): Zone [/Uin] on F.Cu

DRC failed. See report above. No files produced
```

These are precisely the errors reported by KiCAD GUI. Could you try it with the upstream version of KiKit (though there were no changes regarding this IIRC) or possibly with the RC candidate for KiCAD 8.0.1 (which contains a ton of bug fixes).

greenscreenflicker, 2024-03-10:
@yaqwsx now mark as ignore and you should have issue #648

greenscreenflicker, 2024-03-10:
@yaqwsx yes, I'll try to do it tomorrow in the office.

yaqwsx, 2024-03-10:
Ok, there seems to be some mismatch between #654 and #648 (both in the description and steps to reproduce).

96a9646 fixes handling of `footprint_type_mismatch` exclusion, which yields `Unhashable type`. I haven't found any other problem with KiKit not treating exclusion right. If you have a reproducible example, please, let me know.

greenscreenflicker, 2024-03-10:
Thank you, will try head in the Office tomorrow morning

yaqwsx, 2024-03-15:
I consider this as fixed. If there are further problem, please, open a new issue with precise steps to reproduce the problem.

closed 2024-03-15 by yaqwsx