yaqwsx/KiKit #656: Footprint text is incorrectly rotated when 'Keep Upright' is set and footprint is rotated

cbull-86, 2024-03-08:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.0 (older version are not supported)

### KiKit version

1.5.0

### KiCAD version

8

### Operating system

Windows

### Description

Footprint text is incorrectly rotated after panelization for some footprints. I have tracked it down to the following criteria:
-Footprint is rotated
-'Keep upright' is set on the text
-Text is rotated
It seems to occur in panelize.py around line 1011 - in a debug session if I manually re-set the KeepUpright after `item.SetKeepUpright(False)` and before `item.SetTextAngle(actualOrientation - footprint.GetOrientation())` the text retains the correct rotation. Likewise the problem is solved by unticking 'keep upright' for the text in KiCad before running panelization.

### Steps to Reproduce

- Place a footprint & rotate
- rotate the footprint text
- ensure 'Keep upright' is ticked on the footprint text
- panelize

yaqwsx, 2024-03-10:
This issue is confirmed; it happens only on KiCAD 8, not KiCAD 7. It seems that the meaning of `TextAngle` has changed between versions:

What happens for KiCAD 7:
```
actualOrientation: 90.0, footprint.GetOrientation(): 90.0
Text angle: 180.0 
New text angle: 0.0
```

What happens for KiCAD 8:
```
actualOrientation: 90.0, footprint.GetOrientation(): 90.0
Text angle: -90.0 
New text angle: 0.0
```

This needs to be investigated further.

yaqwsx, 2024-03-13:
This should be fixed by 495e55c

closed 2024-03-13 by yaqwsx