yaqwsx/KiKit #663: kikit panelize fails on edge cut in footprint: Board rotation has to be passed as EDA_ANGLE, not a number

idank, 2024-03-13:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.0 (older version are not supported)

### KiKit version

python -m kikit.ui, version 1.5.0

### KiCAD version

8.0

### Operating system

Mac

### Description

kikit seems to have an issue when it sees an edge cut inside of a footprint. I went over https://github.com/yaqwsx/KiKit/issues/189 but it seems there's something different here.

To simplify things I used one of the example commands and I've dumbed down my PCB to just the faulty footprint (an LED) surrounded by a rectangle:

[image]

```
% kikit panelize \
    --layout 'grid; rows: 2; cols: 2; space: 2mm' \
    --tabs 'fixed; width: 3mm' \
    --cuts 'mousebites; drill: 0.5mm; spacing: 1mm; offset: 0.2mm; prolong: 0.5mm' \
    --post 'millradius: 1mm' --debug 'trace: true;' \
test2.kicad_pcb panel.kicad_pcb
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
An error occurred: Board rotation has to be passed as EDA_ANGLE, not a number
No output files produced
Traceback (most recent call last):
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize_ui.py", line 217, in panelize
    doPanelization(input, output, preset, plugin)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize_ui.py", line 267, in doPanelization
    ki.buildLayout(preset, panel, input, sourceArea)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize_ui_impl.py", line 255, in buildLayout
    substrates = panel.makeGrid(
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize.py", line 1277, in makeGrid
    boardSize = self.appendBoard(
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize.py", line 950, in appendBoard
    raise RuntimeError("Board rotation has to be passed as EDA_ANGLE, not a number")
RuntimeError: Board rotation has to be passed as EDA_ANGLE, not a number
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'std::vector< ZONE * > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
```

Thank you for this great piece of software.

### Steps to Reproduce

See example file at https://gist.github.com/idank/4b257632136e5c33a13bcc4352e76e5b.

yaqwsx, 2024-03-13:
I think this is fixed in the upstream version (https://github.com/yaqwsx/KiKit/commit/49363ea96f1c1e2947aa3b0125487e35c7311a6b). Could you try it?

idank, 2024-03-14:
Still failing:

```
% kikit --version
python -m kikit.ui, version 1.5.0+10.g49363ea
%  kikit panelize \
    --layout 'grid; rows: 2; cols: 2; space: 2mm' \
    --tabs 'fixed; width: 3mm' \
    --cuts 'mousebites; drill: 0.5mm; spacing: 1mm; offset: 0.2mm; prolong: 0.5mm' \
    --post 'millradius: 1mm' --debug 'trace: true;' \
test2.kicad_pcb panel.kicad_pcb
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
An error occurred: Board rotation has to be passed as EDA_ANGLE, not a number
No output files produced
Traceback (most recent call last):
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize_ui.py", line 217, in panelize
    doPanelization(input, output, preset, plugin)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize_ui.py", line 267, in doPanelization
    ki.buildLayout(preset, panel, input, sourceArea)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize_ui_impl.py", line 255, in buildLayout
    substrates = panel.makeGrid(
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize.py", line 1275, in makeGrid
    boardSize = self.appendBoard(
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize.py", line 948, in appendBoard
    raise RuntimeError("Board rotation has to be passed as EDA_ANGLE, not a number")
RuntimeError: Board rotation has to be passed as EDA_ANGLE, not a number
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'BOX2< VECTOR2< int > > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'VECTOR2< int > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'std::vector< ZONE * > *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
swig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
```

yaqwsx, 2024-03-15:
Unfortunately, I struggle to reproduce the issue on Linux. And I don't own a Mac. I assume this might be related to the numerous warnings from the SWIG wrapper in KiCAD API. 

I will try to find a way of reproducing it, however, it might take a while.

idank, 2024-03-15:
I'd love to use kikit on Linux, but Kicad 8 botched Python's packaging and installing the nightly version didn't help.

yaqwsx, 2024-03-15:
If you are on Debian-based distribution, the packaging is already fixed in testing builds of v8 (this how we use it for CI): https://github.com/yaqwsx/KiKit/blob/495e55c051bb5cb97a580f73d0a5422af7349a74/.github/actions/setup-kicad/setup.sh#L53-L65

idank, 2024-03-15:
Yeah I tried that, for some reason I'm not getting the `/usr/lib/kicad` directory.

As a matter of fact, I do have the file at `/usr/lib/python3/dist-packages/pcbnew.py`, but I didn't want to install kikit system wide as the instructions suggest, so I used pipx and that installed pcbnew in a virtual environment and got the file at `/home/idank/.local/pipx/venvs/kikit/lib/python3.11/site-packages/pcbnewTransition/pcbnew.py` as well.

I'm not sure how kikit is supposed to be installed as simply doing `pip3 install kikit` doesn't work anymore.

yaqwsx, 2024-03-15:
`pip3 install kikit` work on Ubuntu 22.04 with KiCAD 7 and KiCAD 8.0.1-rc2 installed from repositories. No extra steps are needed.

idank, 2024-03-15:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 23.10
Release:	23.10
Codename:	mantic
$ pip3 install kikit
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

yaqwsx, 2024-03-16:
Ah, you mean this change in recent pip. Due to the way KiCAD handles Python plugins, the recommended way is `--break-system-packages` (it is the same as what pip did before this warning appeared).

I have plans for other installation methods, but it is laborious to get working properly (as it has to work on Linux, Flatpak, Windows, and mac), which makes it annoying to test properly.

idank, 2024-03-17:
That does work but I hope you'll figure this part out in the future!

Also, I'm not getting the crashes on Linux so your suspicion that it's Mac related seem correct.

yaqwsx, 2025-01-26:
Could we retry this with the upcoming KiCAD 9?