yaqwsx/KiKit #666: Error: "Unknown file type"

bradanlane, 2024-03-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.0 (older version are not supported)

### KiKit version

1.5.1

### KiCAD version

8.0.1

### Operating system

MacOS 14.4

### Description

Run Kikit to make panel from project.

Onu output is:
```
An error occurred: Unknown file type
No output files produced
```

When I look in the output folder, I see the following files:
```
           13 Mar 18 11:56 panel.kicad_dru
1969916 Mar 18 11:56 panel.kicad_pcb?
       1271 Mar 18 11:56 panel.kicad_prl
   20393 Mar 18 11:56 panel.kicad_pro
```

If I remove the `?` from `panel.kicad_pcb?` then I can open it and it *looks* OK.

### Steps to Reproduce

Here is the kaki JSON:
``` JSON
{
    "layout": {
        "type": "grid",
        "rows": 1,
        "cols": 1,
        "hspace": "2mm",
        "vspace": "2mm"
    },
    "source": {
        "type": "auto",
        "tolerance": "1mm"
    },
    "tabs": {
        "type": "corner",
		"width": "4mm",
        "source": "none"
    },
	"cuts": {
		"type": "mousebites",
		"drill": "0.5mm",
		"spacing": "0.75mm",
		"offset": "-0.35mm",
		"prolong": "-0.4mm"
	},
    "framing": {
        "type": "tightframe",
		"hspace": "5mm",
		"vspace": "5mm",
		"slotwidth": "2.5mm",
        "width": "2.5mm"
    },
	"tooling": {
		"type": "4hole",
		"hoffset": "3.5mm",
		"voffset": "3.5mm",
		"size": "1.8mm"
	},
	"fiducials": {
		"type": "4fid",
		"paste": true,
		"hoffset": "13.5mm",
		"voffset": "3.5mm",
		"coppersize": "1mm",
		"opening": "1.5mm"
	},
	"post": {
		"type": "auto",
		"millradius": "1mm",
		"origin": "bl",
        "copperFill": false
	}
}
```

I will need to redact the pubnew file before I can share it.

yaqwsx, 2024-03-22:
Please rerun it at least with `--debug 'trace: true'` (as requested in the issue template). If you want, you can share the files confidentially via e-mail.

bradanlane, 2024-03-22:
I was getting an error that `--debug` was not a command. I just realized that it doesn't work when using a json file for the configuration. adding the following gave me the output:
```
"debug: {
    "trace": true
}'
```

The results are:
```
An error occurred: Unknown file type
No output files produced
Traceback (most recent call last):
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize_ui.py", line 217, in panelize
    doPanelization(input, output, preset, plugin)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize_ui.py", line 301, in doPanelization
    panel.save(reconstructArcs=preset["post"]["reconstructarcs"],
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/kikit/panelize.py", line 537, in save
    fillBoard = pcbnew.LoadBoard(self.filename)
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pcbnew.py", line 9305, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unknown file type
```

bradanlane, 2024-03-22:
I am also sing a couple other errors but I think they are "downstream" to this one and may go away one I have this issue sorted.

bradanlane, 2024-03-22:
DOH! The issue was the bash file I was using was originally written from a windows machine and had a hidden LF at the end. I was running the bash file from a Mac.

closed 2024-03-22 by bradanlane