yaqwsx/KiKit #667: Custom cuts are not exported

andy9a9, 2024-03-23:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.0 (older version are not supported)

### KiKit version

python -m kikit.ui, version 1.5.1

### KiCAD version

8.0.1

### Operating system

Mac

### Description

Custom cuts are not exported, only holes.

### Steps to Reproduce

Create custom line cuts like on picture.
[image]
Only red-selected circular hole has been "exported", the cuts above not.

yaqwsx, 2024-03-23:
I am sorry, but for such a poor bug report, I can only guess what could have gone wrong. You deliberately ignored including reproducible steps for reproduction ­– you didn't provide the KiKit command nor the source files. I can only guess what you mean by "export". Is it manufacturing data export?

If by "export" you mean panelize, my bet would be that you specified milling simulation but the milled slots are actually smaller than the mill diameter.

andy9a9, 2024-03-23:
Sorry for not adding reproducible steps. I was guessing, that you tell me, this feature is not supported, or ... . Nevertheless I'm adding all required files for reproducing. I removed all parts and keep there only a shafts, so hopefully it's enough.
[kikit-test.zip](https://github.com/yaqwsx/KiKit/files/14733353/kikit-test.zip)

yaqwsx, 2024-03-23:
Well, as I said above - you instruct KiKit to simulate milling with a 1mm endmill (`millradius: 0.5mm`), but your slots and holes are only 0.762 mm wide (see screenshot below). So, KiKit is doing the job like it was expected. It shows you that there will be no slots if you manufacture the PCB with a 1mm endmill.

[image]

I guess you either:
- want to choose a realistic endmill size or
- you are not interested in the simulation, thus, you shouldn't use `millradius`.

closed 2024-03-23 by yaqwsx