yaqwsx/KiKit #675: Kikit not listing layer names in VCuts

CreativeRobotics, 2024-04-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.0 (older version are not supported)

### KiKit version

1.5.1

### KiCAD version

8.0.1

### Operating system

Windows 10

### Description

In the option for VCuts, the layer option does not list layer names, just layer numbers from 0-58. It defaults to layer 0 (Top copper) but I can't identify what layer number corresponds to Edge.Cuts

### Steps to Reproduce

Clean installed KiCad 8.0.1 and Kikit as per Kikit instal instructions. Open standalone PCB Editor, open Kikit pannelisation tool.

yaqwsx, 2024-04-18:
Could you be more specific about the instructions for reproduction and, e.g., provide a screenshot? I just
- opened Pcbnew,
- clicked the penalization GUI icon,
- choose cuts as vctus
- I observe layers listed by name.

[image]

CreativeRobotics, 2024-04-18:
[image]
I recently reinstalled Kicad from scratch because the previous install of version 8.0.1 resulted in Kikit and other plugins not working at all. Everything seemed to be OK with the installation until I tried to make up some pannels with Kikit.

yaqwsx, 2024-04-18:
Ok, it could be Windows-related.

CreativeRobotics, 2024-04-18:
OK, just quit Kicad and its command prompt and restarted the command prompt - now 'kikit --help' works - so is this some weird issue to do with windows and possibbly the one drive?

I will reinstall the front end of Kikit and see what happens.

CreativeRobotics, 2024-04-18:
Ok. I have reinstalled Kikit fully, and the same problem exists. I don't know if this is a windows thing or not, but I never had any issues with Kikit and Kicad before I updated to Kicad 8.

mikhailmihalkov, 2024-04-27:
\OneDrive\ try not to use network and cloud solutions for file location

FractalEngineer, 2024-04-29:
Came to post about it, same issue here; layers show as numbers, but also even when selecting the right number it won't render the V-Cut lines.
Kikit 1.5.1, Kikad 8.0.2, Win11
[image]

pajasry, 2024-05-01:
This issue is not windows related. I am experiencing same behavior with KiCad 7.0 and Fedora 40. I got the correct number from .kicad_pcb file (44 for Edge.Cuts), try panelize and got "Cannot perform:

Error in section cuts: '44'"

CreativeRobotics, 2024-05-01:
> \OneDrive\ try not to use network and cloud solutions for file location

How can I avoid it when this is the default install for Kicad on Windows 10?

FractalEngineer, 2024-05-01:
> This issue is not windows related. I am experiencing same behavior with KiCad 7.0 and Fedora 40. I got the correct number from .kicad_pcb file (44 for Edge.Cuts), try panelize and got "Cannot perform: Error in section cuts: '44'" message.

Might be; I've just tried on the Linux version of Kicad 8.0.2 under WSL and while it **does** list the layers by their names there, it still won't render the V-Cuts on any of the selected layers.

[image]

yaqwsx, 2024-05-02:
This should be fixed in 551e670.

closed 2024-05-02 by yaqwsx

gwideman, 2024-07-12:
I just installed the "current" version 1.5.1 of Kikit today (July 12 2024), and this fix is not in it.  I don't know if that's intentional, but the fix is from over 2 months ago, so I was a little surprised.
I guess this is because the doc-recommended command:
pip install kikit
... relies on pip fetching kikit from Pipy, where the latest version of Kikit is indeed 1.5.1, of March 15th, which does indeed predate this fix.

yaqwsx, 2024-07-13:
You have to install the upstream version. The fix is not present in a released version yet (there should be release in a week or so).