yaqwsx/KiKit #678: Aborted (core dumped)

thisisgregm, 2024-04-27:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.0 (older version are not supported)

### KiKit version

1.5.1

### KiCAD version

7.0.11

### Operating system

Xubuntu 23.10

### Description

I set settings in GUI, it generates a panel a couple of times (changed setting s a few time to make a panel correct), then hanging / shutting the whole KiCAD down.
Same settings used from a terminal (see the command below) give me " Aborted (core dumped)" error message.

### Steps to Reproduce

I use below command (generated by KiKit GUI). It gives me the "Aborted (core dumped)" error on a few PCBs I tried, even some very simple ones. Hopefully that will reproduce the issue to other people too:

```
kikit panelize
     --layout 'hspace: 0.5mm; vspace: 0.5mm; hbackbone: 2.5mm; vbackbone: 2.5mm; rows: 5; cols: 2'
     --tabs 'vwidth: 5mm; hwidth: 5mm; spacing: 20mm'
     --cuts 'type: vcuts; layer: 43'
     --framing 'type: frame; hspace: 0.5mm; vspace: 0.5mm; width: 7mm; mintotalheight: 70mm; mintotalwidth: 70mm'
     --fiducials 'type: 3fid; hoffset: 3.35mm; voffset: 3.35mm; opening: 2mm'
     --text 'type: simple; hoffset: 10mm; voffset: 3mm; hjustify: left; layer: 37; text: PCB-123 TITLE (YYWW: 2419); anchor: tl'
     '/path/to/project/test-pcb.kicad_pcb' panel.kicad_pcb
Aborted (core dumped)
```

Many thanks.

yaqwsx, 2024-04-29:
I am afraid that without a stack trace, there is not much we can do. I have tried reproducing the issue without success so far.

thisisgregm, 2024-04-30:
Thanks for responding. Is this helpful:

```
coredumpctl gdb -1
           PID: 57864 (kikit)
           UID: 1000 (myuser)
           GID: 1000 (myuser)
        Signal: 6 (ABRT)
     Timestamp: Tue 2024-04-30 11:04:35 BST (5s ago)
  Command Line: /usr/bin/python3 /home/myuser/.local/bin/kikit panelize --layout $'hspace: 0.5mm; vspace: 0.5mm; hbackbone: 2.5mm; vbackbone: 2.5mm; rows: 5; cols: 2' --tabs $'vwidth: 5mm; hwidth: 5mm; spacing: 20mm' --cuts $'type: vcuts; layer: 43' --framing $'type: frame; hspace: 0.5mm; vspace: 0.5mm; width: 7mm; mintotalheight: 70mm; mintotalwidth: 70mm' --fiducials $'type: 3fid; hoffset: 3.35mm; voffset: 3.35mm; opening: 2mm' --text $'type: simple; hoffset: 10mm; voffset: 3mm; hjustify: left; layer: 37; text: PCB-123 TITLE (YYWW: 2419); anchor: tl' /home/myuser/Documents/kicad/Projects/test/test.kicad_pcb panel.kicad_pcb
    Executable: /usr/bin/python3.11
 Control Group: /user.slice/user-1000.slice/session-c1.scope
          Unit: session-c1.scope
         Slice: user-1000.slice
       Session: c1
     Owner UID: 1000 (myuser)
       Boot ID: 1f187ef2998f4a8dbada8fb1c6f7c579
    Machine ID: 0b93f6a03a7e4a6a9b8626527b455e0f
      Hostname: myuser-Inspiron-7501
       Storage: /var/lib/systemd/coredump/core.kikit.1000.1f187ef2998f4a8dbada8fb1c6f7c579.57864.1714471475000000.zst (present)
  Size on Disk: 24.8M
       Message: Process 57864 (kikit) of user 1000 dumped core.
                
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely.libs/libgeos-f99bd675.so.3.11.3 without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely.libs/libgeos-f99bd675.so.3.11.3
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy.libs/libquadmath-96973f99.so.0.0.0 without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy.libs/libquadmath-96973f99.so.0.0.0
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy.libs/libgfortran-040039e1.so.5.0.0 without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy.libs/libgfortran-040039e1.so.5.0.0
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/random/_generator.cpython-311-x86_64-linux-gnu.so from deb systemd-253.5-1ubuntu6.1.amd64
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy.libs/libopenblas64_p-r0-0cf96a72.3.23.dev.so without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy.libs/libopenblas64_p-r0-0cf96a72.3.23.dev.so
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely.libs/libgeos_c-da48124d.so.1.17.3 without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely.libs/libgeos_c-da48124d.so.1.17.3
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely/lib.cpython-311-x86_64-linux-gnu.so without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely/lib.cpython-311-x86_64-linux-gnu.so
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/random/_sfc64.cpython-311-x86_64-linux-gnu.so from deb systemd-253.5-1ubuntu6.1.amd64
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/random/_pcg64.cpython-311-x86_64-linux-gnu.so from deb systemd-253.5-1ubuntu6.1.amd64
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/random/_philox.cpython-311-x86_64-linux-gnu.so from deb systemd-253.5-1ubuntu6.1.amd64
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/random/_mt19937.cpython-311-x86_64-linux-gnu.so from deb systemd-253.5-1ubuntu6.1.amd64
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/random/_bounded_integers.cpython-311-x86_64-linux-gnu.so from deb systemd-253.5-1ubuntu6.1.amd64
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/random/_common.cpython-311-x86_64-linux-gnu.so from deb systemd-253.5-1ubuntu6.1.amd64
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/core/_multiarray_umath.cpython-311-x86_64-linux-gnu.so without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/core/_multiarray_umath.cpython-311-x86_64-linux-gnu.so
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/random/bit_generator.cpython-311-x86_64-linux-gnu.so from deb systemd-253.5-1ubuntu6.1.amd64
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/random/mtrand.cpython-311-x86_64-linux-gnu.so from deb systemd-253.5-1ubuntu6.1.amd64
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/fft/_pocketfft_internal.cpython-311-x86_64-linux-gnu.so from deb systemd-253.5-1ubuntu6.1.amd64
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/linalg/_umath_linalg.cpython-311-x86_64-linux-gnu.so without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/linalg/_umath_linalg.cpython-311-x86_64-linux-gnu.so
                Module /home/myuser/.local/lib/python3.11/site-packages/numpy/core/_multiarray_tests.cpython-311-x86_64-linux-gnu.so from deb systemd-253.5-1ubuntu6.1.amd64
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely/_geos.cpython-311-x86_64-linux-gnu.so without build-id.
[... cut ...]
                
                Stack trace of thread 57867:
                #0  0x00007cdc4e8940f6 __futex_abstimed_wait_common64 (libc.so.6 + 0x940f6)
                #1  0x00007cdc4e896a7e __pthread_cond_wait_common (libc.so.6 + 0x96a7e)
                #2  0x00007cdc409506fb n/a (/home/myuser/.local/lib/python3.11/site-packages/numpy.libs/libopenblas64_p-r0-0cf96a72.3.23.dev.so + 0x3506fb)
                #3  0x0000000000000002 n/a (n/a + 0x0)
                ELF object binary architecture: AMD x86-64
```

thisisgregm, 2024-05-02:
Tried it today in Virtualbox on Xubuntu 24.04, Python 3.12 and KiCAD 8.0.2 - the same issue even on the simplest PCB (small rectangular with 2 header connectors and 2 SMT resistors).

thisisgregm, 2024-05-04:
I tried GUI version and also Aborted / core dumped (this time generated by pcbnew):

```
           PID: 181010 (pcbnew)
           UID: 1000 (myuser)
           GID: 1000 (myuser)
        Signal: 6 (ABRT)
     Timestamp: Sat 2024-05-04 18:56:53 BST (1min 50s ago)
  Command Line: pcbnew
    Executable: /usr/bin/pcbnew
 Control Group: /user.slice/user-1000.slice/session-c1.scope
          Unit: session-c1.scope
         Slice: user-1000.slice
       Session: c1
     Owner UID: 1000 (myuser)
       Boot ID: 79b5ff1aec0c4325a985c88d708d731d
    Machine ID: 0b93f6a03a7e4a6a9b8626527b455e0f
      Hostname: myuser-Inspiron-7501
       Storage: /var/lib/systemd/coredump/core.pcbnew.1000.79b5ff1aec0c4325a985c88d708d731d.181010.1714845413000000.zst (present)
  Size on Disk: 40.7M
       Message: Process 181010 (pcbnew) of user 1000 dumped core.
                
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely.libs/libgeos-f99bd675.so.3.11.3 without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely.libs/libgeos-f99bd675.so.3.11.3
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely/_geometry_helpers.cpython-311-x86_64-linux-gnu.so without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely/_geometry_helpers.cpython-311-x86_64-linux-gnu.so
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely.libs/libgeos_c-da48124d.so.1.17.3 without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely.libs/libgeos_c-da48124d.so.1.17.3
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely/_geos.cpython-311-x86_64-linux-gnu.so without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely/_geos.cpython-311-x86_64-linux-gnu.so
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely/lib.cpython-311-x86_64-linux-gnu.so without build-id.
                Module /home/myuser/.local/lib/python3.11/site-packages/shapely/lib.cpython-311-x86_64-linux-gnu.so
                Module libsystemd.so.0 from deb systemd-253.5-1ubuntu6.1.amd64
                Stack trace of thread 181288:
                #0  0x00007ec96f099a1b __pthread_kill_implementation (libc.so.6 + 0x99a1b)
                #1  0x00007ec96f0428e6 __GI_raise (libc.so.6 + 0x428e6)
                #2  0x00007ec96f0268b7 __GI_abort (libc.so.6 + 0x268b7)
                #3  0x00007ec96f027750 __libc_message (libc.so.6 + 0x27750)
                #4  0x00007ec96f0a3ce7 malloc_printerr (libc.so.6 + 0xa3ce7)
                #5  0x00007ec96f0a6109 _int_free (libc.so.6 + 0xa6109)
                #6  0x00007ec96f0a8873 __GI___libc_free (libc.so.6 + 0xa8873)
                #7  0x00007ec969cb68a6 n/a (_pcbnew.kiface + 0xab68a6)
                #8  0x00007ec969c5c8eb n/a (_pcbnew.kiface + 0xa5c8eb)
                #9  0x00007ec96f9e0a21 n/a (libpython3.11.so.1.0 + 0x1e0a21)
                #10 0x00007ec96fa86a28 n/a (libpython3.11.so.1.0 + 0x286a28)
                #11 0x00007ec96f90ef28 _PyEval_EvalFrameDefault (libpython3.11.so.1.0 + 0x10ef28)
                #12 0x00007ec96fa67452 n/a (libpython3.11.so.1.0 + 0x267452)
                #13 0x00007ec96f90b41e _PyEval_EvalFrameDefault (libpython3.11.so.1.0 + 0x10b41e)
                #14 0x00007ec96fa67452 n/a (libpython3.11.so.1.0 + 0x267452)
                #15 0x00007ec96f97bd6c n/a (libpython3.11.so.1.0 + 0x17bd6c)
                #16 0x00007ec96fb9eb72 n/a (libpython3.11.so.1.0 + 0x39eb72)
                #17 0x00007ec96fac036b n/a (libpython3.11.so.1.0 + 0x2c036b)
                #18 0x00007ec96f097b5a start_thread (libc.so.6 + 0x97b5a)
                #19 0x00007ec96f1285fc __clone3 (libc.so.6 + 0x1285fc)
                
                Stack trace of thread 181010:
                #0  0x00007ec96f0940f6 __futex_abstimed_wait_common64 (libc.so.6 + 0x940f6)
                #1  0x00007ec96f09f076 __new_sem_wait_slow64 (libc.so.6 + 0x9f076)
                #2  0x00007ec96fac865d PyThread_acquire_lock_timed (libpython3.11.so.1.0 + 0x2c865d)
                #3  0x00007ec96fb9da53 n/a (libpython3.11.so.1.0 + 0x39da53)
                #4  0x00007ec96fb9db9b n/a (libpython3.11.so.1.0 + 0x39db9b)
                #5  0x00007ec96f9851ae n/a (libpython3.11.so.1.0 + 0x1851ae)
                #6  0x00007ec96f979cb7 PyObject_Vectorcall (libpython3.11.so.1.0 + 0x179cb7)
                #7  0x00007ec96f90acdf _PyEval_EvalFrameDefault (libpython3.11.so.1.0 + 0x10acdf)
                #8  0x00007ec96fa67452 n/a (libpython3.11.so.1.0 + 0x267452)
                #9  0x00007ec96f97bc48 n/a (libpython3.11.so.1.0 + 0x17bc48)
                #10 0x00007ec9496b3a88 _ZN12wxPyCallback12EventThunkerER7wxEvent (_core.cpython-311-x86_64-linux-gnu.so + 0x4b3a88)
                #11 0x00007ec9715a76e2 _ZN12wxEvtHandler23ProcessEventIfMatchesIdERK21wxEventTableEntryBasePS_R7wxEvent (libwx_baseu-3.2.so.0 + 0x1a76e2)
                #12 0x00007ec9715a9d9e _ZN12wxEvtHandler23SearchDynamicEventTableER7wxEvent (libwx_baseu-3.2.so.0 + 0x1a9d9e)
                #13 0x00007ec9715aa104 _ZN12wxEvtHandler11TryHereOnlyER7wxEvent (libwx_baseu-3.2.so.0 + 0x1aa104)
[... 340 lines cut ...]
[New LWP 181141]
[New LWP 181122]
[New LWP 181027]
[New LWP 181128]
[New LWP 181023]
[New LWP 181026]
[New LWP 181130]
[New LWP 181030]
[New LWP 181121]
[New LWP 181131]
[New LWP 181129]
[New LWP 181132]
[New LWP 181289]
[New LWP 181024]
[New LWP 181124]
[New LWP 181140]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `pcbnew'.
Program terminated with signal SIGABRT, Aborted.
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimised out>)
    at ./nptl/pthread_kill.c:44
[Current thread is 1 (Thread 0x7ec94b70b6c0 (LWP 181288))]

```

yaqwsx, 2024-05-04:
Could you install package kicad-dbg? All the relevant parts of the stack trace are N/A.

thisisgregm, 2024-05-04:
OK, will do and try again.

I've done some testing.

All is good until I use text with anchor **_tl_**. Then changed the anchor back to mt, then also tried tr and bl. And all is fine.
Seems like it only crashes when I use anchor tl.

Could you please confirm from your site it this is also happening to you the same way?

yaqwsx, 2025-01-26:
I am closing this issue as we are about to drop the support for KiCAD v7 in a week or so.

closed 2025-01-26 by yaqwsx