yaqwsx/KiKit #681: Cannot Perform Error

vardelean, 2024-05-08:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.0 (older version are not supported)

### KiKit version

1.5.1

### KiCAD version

8.0.2

### Operating system

Windows 11

### Description

When trying to panelize, I get the following error:
 
[image]

### Steps to Reproduce

1) Open a new standalone PCB editor
2) Open the KiKit  plugin from the new editor window
3) Select pcb input file that I want to panelize
4) Run Panelize
5) The error pops up

Things I tried:
- Downgraded to KiCad Version 8.0.1, no change
- Uninstalled the KiKit plugin using KiCad's Plugin and Content Manager and then reinstalled KiKit
- Note that the Plugin and Content Manager shows KiKit version 1.5.0, but what is installed after pip install is version 1.5.1, see below:
kikit --version
kikit, version 1.5.1

Any help is much, much appreciated!
Thank you!

yaqwsx, 2024-05-12:
Without a PCB file that we can use for reproduction, there is not much we can do...

vardelean, 2024-05-12:
No problem, please see attached.

Thank you!
[Boost_Max_1A.zip](https://github.com/yaqwsx/KiKit/files/15287647/Boost_Max_1A.zip)

yaqwsx, 2024-05-13:
Well, there is no board in the files you provided.

vardelean, 2024-05-13:
I apologize, I zipped the wrong file, my bad!
Please see attached.
Thank you!

[Boost_Max_1A_Board.zip](https://github.com/yaqwsx/KiKit/files/15300875/Boost_Max_1A_Board.zip)

yaqwsx, 2024-05-14:
I just tested this on KiKit 1.5.1+16.g5a5d509 and I can't reproduce the issue. Are you sure there are no extra steps to reproduce this issue?

vardelean, 2024-05-14:
Thank you for looking into this! 

The steps I highlighted initially are the only ones I took: open a new PCB Editor (standalone), open KiKit plugin, select the PCB file, select a Json file I already saved, and hit panelize.  

I believe the issue may be with my computer or with the installation of the KiKIt plugin, because I get the same error on any PCBs I select for panelization, not just the one I sent you. It just happened I noticed the issue on the PCB I attached earlier because I never used the KiKit plugin before.

As I said before, I tried to uninstall KiKit using the KiCad Component Manager and then re-install KiKIt, but that did not help.
I also tried to turn off the Firewall (Avast), did not help either.  

Any suggestions are very much appreciated.

Thank you!

yaqwsx, 2024-05-15:
You specified selecting the board via footprint annotation in the section `source` but you didn't provide the annotation reference. Therefore, the error is expected, it should just have more clear wording. The wording was fixed in 32187ad. If you change source to `auto`, everything works.

closed 2024-05-15 by yaqwsx

vardelean, 2024-05-15:
This is wonderful, thank you so much for your help! Works like a charm!