yaqwsx/KiKit #683: Reference designators are wrongly rotated

greenscreenflicker, 2024-05-10:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.0 (older version are not supported)

### KiKit version

kikit, version 1.5.0

### KiCAD version

8.0.1

### Operating system

Windows 11

### Description

Sometimes reference designators are rotated differently than in the original pcb. It seems to be a newer issue.

### Steps to Reproduce

Make a pcb with several reference designators on it, panalize it (with roatation) and observe wrongl rotated reference designators.

kinoshita-lab, 2024-05-11:
@greenscreenflicker 
Same here, it's already fixed for v1.5.1.
I have just confirmed it works for me, also using Windows 11.

Please try:
https://github.com/yaqwsx/KiKit/releases/tag/v1.5.1

yaqwsx, 2024-05-12:
Duplicate of #679 and #656. Already fixed in v1.5.1. Closing.

closed 2024-05-12 by yaqwsx