yaqwsx/KiKit #686: DRC errors are not ignored

greenscreenflicker, 2024-05-13:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

kikit, version 1.5.1+16.g5a5d509

### KiCAD version

Application: KiCad PCB Editor x64 on x64  Version: 8.0.1, release build  Libraries: 	wxWidgets 3.2.4 	FreeType 2.12.1 	HarfBuzz 8.3.0 	FontConfig 2.14.2 	libcurl/8.5.0-DEV Schannel zlib/1.3  Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW  Build Info: 	Date: Mar 15 2024 01:52:47 	wxWidgets: 3.2.4 (wchar_t,wx containers) 	Boost: 1.83.0 	OCC: 7.7.1 	Curl: 8.5.0-DEV 	ngspice: 42 	Compiler: Visual C++ 1936 without C++ ABI  Build settings:

### Operating system

Windows 11

### Description

DRC expections are not ignored if marked so in PCBNew

### Steps to Reproduce

Run the PCB as send per email, run drc in pcbnew, observe it runs fine.
Run export command, observe DRC errors.

yaqwsx, 2024-07-19:
This was already addressed - the problem is in KiCAD that saves ignored rules on project close. Until then, the ignore rules are not written to filesystem, hence, KiKit can't read them.

closed 2024-07-19 by yaqwsx

greenscreenflicker, 2024-07-19:
yes, was already addressed. 
It could be nice to make a kicad issue that saves also the ignore rules on PCB safe in PCBnew.