yaqwsx/KiKit #689: `solderMaskMargin` option for `--tooling` doesn't seem to do anything

comps, 2024-05-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

1.5.1

### KiCAD version

7.0.11-2.fc39

### Operating system

Linux, Fedora 39

### Description

`solderMaskMargin` added by https://github.com/yaqwsx/KiKit/pull/634 doesn't seem to do anything, I'm using it as
```
kikit panelize \
        --layout 'grid; rows: 2; cols: 2' \
        --tooling '3hole; hoffset: 2.5mm; voffset: 2.5mm; size: 1.152mm; solderMaskMargin: 1.3mm' \
        test-project/test-project.kicad_pcb \
        temp/temp.kicad_pcb
```
but looking at the generated PCB, the soldermask comes right up to the hole.

I also tried `soldermaskmargin` (lowercase) given the code from https://github.com/yaqwsx/KiKit/pull/634 seems to use both, no effect.

Selecting one of the NPTHs and copying it gives me:
```
(footprint "NPTH" (version 20221018) (generator pcbnew)
  (layer "F.Cu")
  (tstamp 3787a2a3-44f5-4596-b7da-ef09d66d1355)
  (at 0 0)
  (attr through_hole board_only)
  (fp_text reference "KiKit_TO_3" (at 0 0.5) (layer "F.SilkS") hide
      (effects (font (size 1 1) (thickness 0.15)))
    (tstamp 830f0663-20e1-40ec-9675-2674b43d8276)
  )
  (fp_text value "NPTH" (at 0 -0.5) (layer "F.Fab") hide
      (effects (font (size 1 1) (thickness 0.15)))
    (tstamp d51bcfe6-7be4-4354-ad23-1b5f147f997f)
  )
  (pad "" np_thru_hole circle (at 0 0) (size 1.152 1.152) (drill 1.152) (layers "*.Mask") (tstamp c1e517cf-8b31-4232-aa95-29afb8440ce2))
)
```
I can edit the footprint and specify the margin manually, which results in
```
  (pad "" np_thru_hole circle (at 0 0) (size 1.152 1.152) (drill 1.152) (layers "*.Mask")
    (solder_mask_margin 1.3) (tstamp c1e517cf-8b31-4232-aa95-29afb8440ce2))
```
(units probably wrong, KiCad uses mask "expansion", not absolute diameter), but even if `solderMaskMargin` would do this, KiCAD warns that the parameter has no effect on non-copper layers.

[image]
[image]

### Steps to Reproduce

1. Make a trivial board - I've attached one per your policy, but it'll be faster to place one resistor, draw edge rectangle and save
2. Run
```
kikit panelize \
        --layout 'grid; rows: 2; cols: 2' \
        --tooling '3hole; hoffset: 2.5mm; voffset: 2.5mm; size: 1.152mm; solderMaskMargin: 1.3mm' \
        test-project/test-project.kicad_pcb \
        temp/temp.kicad_pcb
```
3. Observe the missing solder mask margin

---
Attached:
* [test-project.kicad_pcb.gz](https://github.com/yaqwsx/KiKit/files/15359825/test-project.kicad_pcb.gz)

yaqwsx, 2024-05-18:
This is a known bug in KiCAD 7 API. It works well with KiCAD 8. At the moment, there is no other work-around other than to switch to KiCAD 8.

comps, 2024-05-18:
Alright, thanks for the info, I'll close this then.

closed 2024-05-18 by comps

thenickdude, 2024-10-08:
I have this problem on KiCAD 8.0.5, kikit 1.6.0, macOS 12 Monterey, and it reproduces there with OP's example command. 

The solder mask opening is the same size as the hole regardless of what value is entered.

baileymontano, 2024-11-04:
I am also still seeing this on KiCAD 8.0.6, kikit 1.6.0 Win 11 x64. I think this is a holdover issue from KiCAD changing how it handles solder mask expansion for footprints that aren't on copper layers.

I believe I can get the desired behavior if I set the pad diameter to the solder mask diameter. Could also set the NPTH to exist on either or both copper layers (though this will add footprint clearances as well), or edit the footprint to add an SMD aperture that exists on the mask layers.