yaqwsx/KiKit #694: kikit crash when trying to panelized PCB

kfirmaymon84, 2024-05-22:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

1.5.1

### KiCAD version

8.0.2

### Operating system

Windows 11

### Description

Hi,
When I run kikit panelize, 
[image]

the "Running kikit widows freeze,
[image]

after a couple of secoud, I got a message "Kicad PCB Editor 8.0.2 is not responding"
[image]

if I close the program and open the PCB file I can see that the file is been penalized.

if I run kikit from CLI everything runs ok,

### Steps to Reproduce

Just penalize PCB.
[EvoAcs_SideTouchSens.zip](https://github.com/yaqwsx/KiKit/files/15402602/EvoAcs_SideTouchSens.zip)

thisisgregm, 2024-06-05:
Could be related to this issue:
https://github.com/yaqwsx/KiKit/issues/678
See my comment here:
https://github.com/yaqwsx/KiKit/issues/678#issuecomment-2094349259
I can see you use **_anchor: tl_**, the same as I did to reproduce my issue. No problems with other options for anchor.

kfirmaymon84, 2024-06-20:
Dident help, 
I tried to do only panelized and it also happened.
[image]

yaqwsx, 2024-07-01:
@kfirmaymon84: Could you try the upstream version? I am able to panelize the board with the upstream version of KiKit. So I would like to rule the possibility this issue is caused by your setup (e.g., Windows or dependencies mismatch).

JeremiahGillis, 2024-07-19:
I am on Windows 11, KiCad 8.0.4, and KiKit 1.6.0. I can confirm that the GUI will lock up and eventually close.

[image]

I switched to the CLI and discovered another issue. The output kicad_pcb file will be blank if text is enabled and the anchor is set to `tl`. The output file works correctly when the anchor is set other points such as `mt` or 'tr'.

I went back to the GUI and turned my text off, but KiCad will still lock up and eventually close.

I also see that KiKit now requires an output file in the GUI. If I set this to the currently opened file since I want the panel to show up like previous KiKit versions, KiCad will crash quickly after the "Running kikit" dialog opens. If I set the output file to something new, not opened, KiCad takes awhile to lockup an crash during panelization.

Since there is an output option in the GUI now, are you moving away from panelizating the currently opened file? IE You want to panelize the current open board?

Here is my JSON file:

```
{
    "layout": {
        "rows": "5",
        "cols": "2"
    },
    "source": {
        "tolerance": "10mm"
    },
    "tabs": {
        "type": "full"
    },
    "cuts": {
        "type": "vcuts"
    },
    "framing": {
        "type": "railslr",
        "hspace": "0mm"
    },
    "tooling": {
        "type": "4hole",
        "hoffset": "2.5mm",
        "voffset": "5mm",
        "size": "2mm"
    },
    "fiducials": {
        "type": "4fid",
        "hoffset": "2.5mm",
        "voffset": "10mm",
        "opening": "2mm"
    },
    "text": {
        "type": "simple",
        "hoffset": "2.5mm",
        "voffset": "140mm",
        "orientation": "90deg",
        "text": "JLCJLCJLCJLC",
        "anchor": "tl"
    },
    "page": {
        "type": "A2",
        "posx": "200mm",
        "posy": "50mm"
    }
}
```

yaqwsx, 2024-07-21:
> Since there is an output option in the GUI now, are you moving away from panelizating the currently opened file? IE You want to panelize the current open board?

No, the whole point of GUI existence is that people can quickly preview the panel and iterate to get the desired shape.  PcbNew is used as a live preview of the panel. When you are satisfied with the result, you should get the penalization JSON and put it in your makefile or CI or another automated pipeline for building production files. GUI was never meant as a way of preparing manufacturing files. Unless something significantly changes in KiCAD, it will never be. KiCAD API is limited, and there are numerous aspects of the PCB we cannot modify via the API (e.g., sheet size, custom DRC rules, rules violation exclusions etc.). Therefore, KiKit parses and modifies the board, project, and schematic files. These changes cannot be brought to the "live view" provided by PcbNew. So when you save the board from preview, all of these features are missing, and the panels are not passing DRC or have other defects.

However, people insisted on saving the preview and using it for production. They were constantly reporting, "Hey, the panel from the GUI does not pass DRC!". Hence, I introduced the output file parameter in GUI in v1.6.0 to make them happy. They can use the GUI, they see live preview and they also get a file that is not damaged.

Back to the original issue, can you confirm that in v1.6.0:
- the file panelizes just fine and in a reasonable time in CLI?
- the panel takes forever/crashes in GUI?

So far, I struggle to reproduce the issue you report on both of my machines running Ubuntu 22.04 and Windows 11. The board panelizes just fine.

Is it possible that you output the panel to a network location? Or to a location handled by OneDrive cloud sync?

JeremiahGillis, 2024-07-21:
> Back to the original issue, can you confirm that in v1.6.0:
> 
> * the file panelizes just fine and in a reasonable time in CLI?
> * the panel takes forever/crashes in GUI?

Yes, I just panelized four projects. My most complex one took 5 seconds. Like I mentioned before, the panel will be blank though if text is enabled and the anchor point is set to `tl`.

KiCad will crash with the exact same JSON file used in the CLI through the GUI.

> Is it possible that you output the panel to a network location? Or to a location handled by OneDrive cloud sync?

My project is local, both the input and out, OneDrive is disabled. My local storage is a PCIe 4.0 NVMe.

I've spent some time this morning to see how I can help you replicate this. The GUI crashes for me with your basic example `conn.kicad_pcb` while using CLI command works fine.

CLI:
```
kikit panelize^
    --layout "rows: 2; cols: 2" ^
    --tabs "type: full" ^
    --cuts "type: vcuts" ^
    "C:\Users\jerem\Downloads\KiKit-master\LocalTest\conn.kicad_pcb" "C:\Users\jerem\Downloads\KiKit-master\LocalTest\panel.kicad_pcb"
```
GUI:
[image]

I have uninstalled KiCad 7.0.10, 8.0.4, and deleted the KiCad folder from my Documents which stores the plugins. After a fresh install of only 8.0.4, installing only the KiKit plug in and the backend, the sample above still crashes in the GUI.

Backend install log:
```
Successfully installed PrettyTable-0.7.2 PyMeta3-0.5.1 click-8.1.7 colorama-0.4.6 commentjson-0.9.0 euclid3-0.1 kikit-1.6.0 lark-parser-0.7.8 markdown2-2.5.0 pcbnewTransition-0.4.1 ply-3.11 pybars3-0.9.7 pypng-0.0.19 shapely-2.0.5 solidpython-1.1.3
```

Windows Info:
```
OS Name: Microsoft Windows 11 Pro
Version: 10.0.22631 Build 22631
Processor: AMD Ryzen 9 5950X 16-Core Processor, 3401 Mhz, 16 Core(s), 32 Logical Processor(s)
Installed Physical Memory (RAM): 64.0 GB
GPU: Nvidia RTX 3090, Driver Version 555.99
```

I have another Windows 11 and Windows 10 computer that I will test on and report back within the next day.

JeremiahGillis, 2024-07-22:
I have the exact same behavior on my Windows 11 laptop. It works just fine in the CLI but not the GUI.

[image]

```
OS Name: Microsoft Windows 11 Pro
Version: 10.0.22631 Build 22631
Processor: 13th Gen Intel(R) Core(TM) i9-13900HX, 2200 Mhz, 24 Core(s), 32 Logical Processor(s)
Installed Physical Memory (RAM): 64.0 GB
GPU: Nvidia RTX 4090, Driver Version 556.12
```

JeremiahGillis, 2024-07-22:
My Windows 10 machine failed the first attempt, but worked every subsequent attempt. I went back to one of my Windows 11 machines for more testing via the GUI.

1. The progress bar will not move and KiCad will crash with your repo `conn.kicad_pcb` file.
2. If the `conn.kicad_pcb` file is opened, a dialog will prompt about different fills from previous KiCad version. If this is acknowledged, then saved for KiCad 8 format, KiKit panelization behavior is different. Now, KiCad will output a panel file correctly, the progress bar will move a little bit, then KiCad will hang and eventually crash. KiCad will not update the open file with a live preview of the panel. The panels settings are the same ones I used in the top of my post yesterday.
3. I went back to Windows 10 following Step 2 and it still panelizes correctly in the GUI. However, if I try one of my private boards with my JSON file, it will act just like Windows 11 in Step 2. It will output a panel file correctly but lock up afterwards.

If it would help, I could create a private repo and invite you. Then you could try a problem layout file with my json config. You would need to delete this files once you were done testing. Just let me know if that would help.

JeVetter, 2024-10-05:
I have the same problem. The GUI is not freezing when my outputfile has another name, then the opened pcb file what is used as a preview . Typically it has the same name as the project file.
I use :
Win 11
Kikat 7.0.7
Kikit 1.6.0

Hyrla, 2024-10-22:
Hello,
Same issue too.
Win 11
Kicad 8.0.5
Kikit 1.6.0

kfirmaymon84, 2024-10-27:
Still having the same problem,
The workaround is to close the software and re-open the panel file.
the panel file is made but the process is not finished.

yaqwsx, 2024-10-29:
There is an experimental branch (https://github.com/yaqwsx/KiKit/tree/ui_stall) that could solve your issue. Could you try installing it via `pip install https://github.com/yaqwsx/KiKit/archive/ui_stall.zip` and test if it addresses this issue?

Hyrla, 2024-10-29:
> There is an experimental branch (https://github.com/yaqwsx/KiKit/tree/ui_stall) that could solve your issue. Could you try installing it via `pip install https://github.com/yaqwsx/KiKit/archive/ui_stall.zip` and test if it addresses this issue?

Sadly even with this version the UI still crashes. Is there something particular to do to install this version appart from running this pip command and restarting kicad?

[image]

yaqwsx, 2024-11-13:
@Hyrla: Just to be sure, do you follow the instructions for using Kikit correctly? That is:
- opening a new window of **Pcbnew** (i.e., PCB editor) - directly from the start menu, not launching KiCAD and working there.
- then, you run the penalization from dialog from this window.

yaqwsx, 2024-11-18:
Hmm, unfortunately, that means that I am out of ideas for the moment. Any change you work on a drive that is a network drive or a drive backed-up by OneDrive?

yaqwsx, 2025-01-27:
@ColinBathe: Please, test the latest upstream version, not the stable one.

ColinBathe, 2025-01-28:
> [@ColinBathe](https://github.com/ColinBathe): Please, test the latest upstream version, not the stable one.

Is this the correct way of using the upstream version?

`pip install https://github.com/yaqwsx/KiKit/archive/master.zip`

If so, it hasn't fixed the issue.

yaqwsx, 2025-01-28:
You can verify that either in the header of GUI or via `kikit --version` in terminal.

Xiuyee-d, 2025-03-23:
This happens when the output file has the same name as the current blank PCB file, so don't do it like that.

kfirmaymon84, 2025-07-03:
> This happens when the output file has the same name as the current blank PCB file, so don't do it like that.

I give it a try and it's still crashing.

kfirmaymon84, 2025-07-03:
> You can verify that either in the header of GUI or via `kikit --version` in terminal.

Can i do something to help resolve this problem, 
I can give you remote access if you want or can i run it in debug mode so i can see were is stuck? 
I'm electronic engineer and software is not my strong side but i can learn if needed. 

I really love kikit.

yaqwsx, 2025-07-03:
@kfirmaymon84 Your original report is from the version 1.5.1. Does the problem persist even with the current upstream version?

kfirmaymon84, 2025-07-03:
> @kfirmaymon84 Your original report is from the version 1.5.1. Does the problem persist even with the current upstream version?
> 

Yes,
I tested it on different computers and it still crash.

PerennialNovice, 2026-10-07:
This might be related to what I encountered:

Kicad 10.0.6; Kikit 1.8.1

CLI runs fine all the time
GUI runs mostly fine when no target file is present - allthough I am not sure the behaviour is as intended:
pcbnew closes and reopens the created file)
Once, it did not reopen though
If a target file exists, on most runs kikit/pcbnew crashes
sometimes the target file is 0kb size, sometimes it gets created fine, sometimes it gets only half created

Worth opening a separate issue for this?
Or is it really related somehow?

nuzbekov, 2026-10-12:
Hi @PerennialNovice,

This looks closely related to what I've been debugging, so I'd say it's the same underlying problem rather than something that needs a separate issue.

I hit the same symptoms on KiCad 10.0.4 / Windows: with an existing target file, pcbnew crashes with an access violation (0xc0000005), and the output is either 0 KB or half-written. The half-written file is what `Panel.save()` produces after its first `Save()`, before the zone refill step, so the crash happens in the middle of `save()` while running inside pcbnew. The same preset through the CLI works every time, which matches your observation.

I opened two PRs for this:

- #934 – refuses to start when the output file is the board currently open in pcbnew. Saving over the live document is the most dangerous case (pcbnew is also asked to replace the content of the board it has open), and it would also explain the "pcbnew closes and reopens the file" behaviour.
- #935 – runs the panelization in a child interpreter, so a native crash becomes an error message instead of taking pcbnew down. A failed child (non-zero exit code, missing or empty output) is never treated as success.

A few questions that would help narrow it down:

1. Is the target file the same board that is open in pcbnew, or a different file?
2. Does the crash still happen if the target file exists but isn't open anywhere?
3. If you're willing to try, installing from https://github.com/nuzbekov/KiKit (branch `all-fixes`, which combines #932–#935) and reporting whether the crashes go away would be a useful independent confirmation.

If you can share the board or the preset, I'm happy to look at it too.

PerennialNovice, 2026-10-12:
Hi @nuzbekov,

on some repeated tests, it seems that the one time that the panelized board got opened in pcbnew was an exceptional event. pcbnew crashed every time, regardless of the existence of the output file

1. only pcbnew standalone open with no board loaded 
2. yes, see 1.
3. will do, but need more instructions since not familiar with building things out of github... sorry

[MP3438.zip](https://github.com/user-attachments/files/33321152/MP3438.zip)

nuzbekov, 2026-10-12:
Thanks for the files. I ran your board and preset on KiCad 10.0.4 / Windows (KiKit 1.8.1): both the CLI and the GUI plugin produce the panel without crashing, so I can't reproduce it on my side. Since you're on 10.0.6, this may be specific to that KiCad version (@buganini above saw a crash in `SETTINGS_MANAGER::SaveProjectAs()` on 10.0.5).

Here is a self-contained way to check whether the pending fixes help in your setup. No building is needed, it's one pip command run with KiCad's own Python. Open "KiCad Command Prompt" from the Start menu (or a normal terminal) and run:

    "C:\Program Files\KiCad\10.0\bin\python.exe" -m pip install --upgrade --force-reinstall --no-deps git+https://github.com/nuzbekov/KiKit.git@all-fixes

(Adjust the path if KiCad is installed elsewhere. `git` must be installed. If it isn't, use this instead:
`... -m pip install --upgrade --force-reinstall --no-deps https://github.com/nuzbekov/KiKit/archive/refs/heads/all-fixes.zip`)

Then restart KiCad and run the KiKit plugin on your board again. This branch combines #932–#935; the relevant part for crashes is #935, which runs the panelization in a separate process, so a pcbnew crash turns into an error dialog with the reason instead of closing the application.

To go back to the released version:

    "C:\Program Files\KiCad\10.0\bin\python.exe" -m pip install --upgrade --force-reinstall --no-deps kikit==1.8.1

Possible outcomes:
- It works now → the fix in #935 addresses your case; please add a comment on #935.
- You get an error dialog instead of a crash → please paste the text here, it shows the real cause.
- It still crashes → please post the "Application Error" entry for pcbnew.exe from Event Viewer (Windows Logs → Application): exception code and faulting module.

That's all from my side. I'm not able to dig further without a reproduction on my machine, but #934/#935 contain the details of what I found.

PerennialNovice, 2026-10-14:
Good morning @nuzbekov,

Sorry for the delay...
Just tested your fixes and both GUI and CLI plugin produce correct output now!
So... works on my machine too ;)

Works with 10.0.6 and 10.0.7

Thanks

nuzbekov, 2026-10-14:
Thanks a lot for testing, and for confirming on both 10.0.6 and 10.0.7. That's very useful, especially since I couldn't reproduce it on my side. The fix is in #935 (together with #934); I'll leave the confirmation there for the maintainer.

PerennialNovice, 2026-10-15:
Glad to be able to help
Not much else I can give back, so happy I can contribute at least somehow

FractalEngineer, 2026-11-12:
> Thanks a lot for testing, and for confirming on both 10.0.6 and 10.0.7. That's very useful, especially since I couldn't reproduce it on my side. The fix is in [...]

Hey mate, had the same issue since kicad 10 some (pcbnew crashing after panelizing) and applying your patch fixed it, thank you for sharing