yaqwsx/KiKit #695: Regression in internal edge cuts, panelize rounds edges

idank, 2024-05-23:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

kikit, version 1.5.1

### KiCAD version

8.0.2

### Operating system

Ubuntu

### Description

panelize seems to go from this:

[image]

to this:

[image]

It used to work OK a couple of months ago when I opened https://github.com/yaqwsx/KiKit/issues/663.

### Steps to Reproduce

https://gist.github.com/idank/e0e9a3a2352c16619a921e15b458fc7a

yaqwsx, 2024-05-23:
You haven't provided the panelization settings. If KiKit processes the file as is, there is no change in the contour outline. I expect you to use `millradius`, but I don't know with what radius. Without this information there is not much I can do.

idank, 2024-05-23:
Ah sorry, I seem to be using:

```
"post": {
        "millradius": "1mm"
    }
```

Was there a recent change so this started to apply to internal edges too? I haven't changed my kikit settings, but there were some kicad minor upgrades since I lasted tested it on this kind of footprint.

yaqwsx, 2024-05-23:
The dogbones in your file are formed by arcs with a radius of 0.5mm. They cannot be milled with an end mill with a radius of 1mm - the mill doesn't fit. Therefore, KiKit's current behavior is correct. If it used to behave differently, it was a bug.

idank, 2024-05-23:
Makes sense, thanks!

closed 2024-05-23 by idank