yaqwsx/KiKit #702: RuntimeError: No board edges found when using PythonAPI

CarbonV, 2024-06-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

1.5.1

### KiCAD version

8.0.3

### Operating system

Windows 11

### Description

I have written a custom script to try and panelize with two boards. I have also run some scripts to print all the layers, Edge.Cuts layer is present and a script that checks for board edges which returns true. I assume I am not using the appendBoard function correctly.

```
from kikit.common import fromMm
import os

# Create a new panel
def create_panel(first_pcb, second_pcb, panel_filename):
    panel = panelize.Panel(panel_filename)

    # Absolute paths to your PCB files
    first_pcb = os.path.abspath(first_pcb)
    second_pcb = os.path.abspath(second_pcb)
    
    # Ensure the PCB files exist
    if not os.path.exists(first_pcb):
        raise FileNotFoundError(f"File not found: {first_pcb}")
    if not os.path.exists(second_pcb):
        raise FileNotFoundError(f"File not found: {second_pcb}")
    
    # Load the boards
    panel.appendBoard(first_pcb, destination=(fromMm(0), fromMm(0)))
    panel.appendBoard(second_pcb, destination=(fromMm(100), fromMm(0)))  # Change positions as needed
    
    # Save the panel
    panel.save(panel_filename)
    print(f"Panel saved as {panel_filename}")

first_pcb = PATH
second_pcb = PATH2
panel_filename = PATH3

print(f"First PCB path: {first_pcb}")
print(f"Second PCB path: {second_pcb}")

create_panel(first_pcb, second_pcb, panel_filename)

```

[image]

[image]

I am running from KiCad commandline.

### Steps to Reproduce

Run code using KiCad commandline.

yaqwsx, 2024-06-18:
Could you try adding a tolerance parameter to `appendBoard`, which is the size of line width of edges?

CarbonV, 2024-06-18:
What is the unit of tolerance? When I use a tolerance of "2" it doesn't make a difference the Line Width of edges is 0.1mm.

yaqwsx, 2024-06-18:
It's the default KiCAD unit - 1nm. There function `fromMm` and `toMm`. You can also import constants `from kikit.units import mm` and then write `1 * mm`.

CarbonV, 2024-06-18:
Adding a tolerance worked! Thank you. Could you elaborate why?

yaqwsx, 2024-06-18:
It seems that some change in the history started to include the line width; we never noticed as CLI always uses some tolerance and also all my custom scripts.

CarbonV, 2024-06-18:
Maybe it's an option to have a default tolerance in panelize.py?