yaqwsx/KiKit #704: Footprint with reference " not found

arslan437, 2024-06-22:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

kikit, version 1.5.1

### KiCAD version

8.0.2

### Operating system

Windows

### Description

I am getting this 

```
foot print with reference " not found
```

I have a found an older issue but that did not helped. 

### Steps to Reproduce

Here is how the error appears 

Open the pcbnew standalone from the start menu
Click on the penalization button. 
selecting the board file. 
than loading the json file with the configuration
```
{
    "layout": {
        "hspace": "2mm",
        "vspace": "2mm",
        "cols": "3"
    },
    "source": {
        "type": "annotation"
    },
    "tabs": {
        "type": "fixed",
        "vcount": "2",
        "fillet": "1mm"
    },
    "cuts": {
        "type": "mousebites",
        "spacing": "1mm",
        "offset": "0.2mm"
    },
    "framing": {
        "type": "railstb"
    },
    "tooling": {
        "type": "4hole",
        "hoffset": "3mm",
        "voffset": "3mm"
    },
    "fiducials": {
        "type": "4fid"
    }
}
```   

And when I click on the penalize button I get this error.

yaqwsx, 2024-06-22:
As the issue template guides, always provide all the files including the board. Othewise, we can only guess what the problem is.

My best guess is this is a duplicate of #681 (did you really search through the issues?): You specified "select board by annotation" but you haven't specified one. The error message is confusing, it will be improved in the next version.

yaqwsx, 2024-06-22:
Error message improved in a8a140573095cdafaa29530dd23f66a7e4c23ec0.

arslan437, 2024-06-23:
Ok, Now I have tried to enter the settings in the fields manually. It worked. I will try replicate the setting from the json file and will update you after that

arslan437, 2024-06-30:
I think I have found the issue.

In the source I have mistaken selected the annotation. The ref is empty. That might have caused the issue. That is my guess, maybe you can check into it. 

[image]

It is working perfectly fine now. Thanks for the prompt response

closed 2024-06-30 by arslan437

yaqwsx, 2024-07-01:
> In the source I have mistaken selected the annotation. The ref is empty. That might have caused the issue. That is my guess, maybe you can check into it.

Is exactly what I said:

> You specified "select board by annotation," but you haven't specified one.