yaqwsx/KiKit #706: numpy 2.0 breaks KiKit, `numpy.lib.utils` has turned into an internal implementation detail

lethalbit, 2024-06-28:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

1.5.1+28.ga8a1405

### KiCAD version

8.0.3

### Operating system

Arch Linux

### Description

When trying to use KiKit on a system that has numpy 2.0.0, it will fail to load due to `numpy.lib.utils` being turned into a private implementation detail, having been renamed `numpy.lib._utils_impl`. (see  [numpy/numpy @ ace4445](https://github.com/numpy/numpy/commit/ace444596424846058d8bfaacace29cd7e0c849a#diff-078f3baf0443b123e856fc9402b17f640136d65b8dec671c3578f90196e92a88))

This causes the panilization section to fail to load taking down the rest of KiCad

### Steps to Reproduce

```
$ python -m venv --system-site-packages .venv
$ source .venv/bin/activate
$ pip install numpy==2.0 kikit
$ python -c 'from kikit.actionPlugins import importAllPlugins; importAllPlugins()'
```

This will throw `ModuleNotFoundError: No module named 'numpy.lib.utils'` from `kikit/actionPlugins/panelize.py` when attempted.

yaqwsx, 2024-06-28:
This should be resolved via eac56af.

closed 2024-06-28 by yaqwsx

stefan-misik, 2024-07-01:
Hi @yaqwsx , do you plan to include this fix in a release soon? This bug is preventing kikit plugin from working on Arch Linux and there is even no up-to-date kikit-git (i.e. nightly build) package in AUR.

yaqwsx, 2024-07-01:
I plan the next release to be on the horizon of 1-2 weeks. If you are in hurry, you should be able to manually update to the upstream version.