yaqwsx/KiKit #710: V-cut railtb doesn't generate both cuts on 0mm offset

a7hybnj2, 2024-07-14:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

./python3.9 kikit --version kikit, version 1.5.1

### KiCAD version

Version: 8.0.1

### Operating system

MacOS: 14.5 (23F79)

### Description

Setting Tooling railtb voffset to 0mm with v-cuts enabled doesn't generate both horizontal v-cuts. You can see in the included screenshots that 0mm doesn't generate the top v-cut, setting 0.5mm generates both.
[image]
[image]

### Steps to Reproduce

Pick any pcb it isn't specific happens on any. Attached is just an edgecut file as I don't want this to be ignored.
[garbage-pcb.zip](https://github.com/user-attachments/files/16227585/garbage-pcb.zip)

```
{
    "layout": {
        "cols": "2"
    },
    "cuts": {
        "type": "vcuts",
        "layer": "Eco1.User"
    },
    "framing": {
        "type": "railstb",
        "vspace": "0mm"
    },
    "tooling": {
        "type": "4hole",
        "hoffset": "5mm",
        "voffset": "2.5mm"
    },
    "fiducials": {
        "type": "4fid",
        "hoffset": "10mm",
        "voffset": "2.5mm",
        "opening": "2mm"
    }
}
```

yaqwsx, 2024-07-15:
This is fixed in 69bcf0b. There was a case of accumulation of numerical error that caused top tabs not being created.

closed 2024-07-15 by yaqwsx

a7hybnj2, 2024-07-17:
@yaqwsx Thanks for taking a look and fixing that so quickly. Looks like the MacOS install script pulls from releases. Will this get baked into new release soon or should I figure out how to pull master somehow?

a7hybnj2, 2024-07-17:
Hey, I saw that it was just a change to substrate so I copied that into my site packages and that seemed to work. Didn't even need to resign. Thanks again!

a7hybnj2, 2024-07-17:
Hey, sorry to be blowing this up but... I just restarted kicad and a cert warning did pop up but I was able to just click "accept modification" and carry on.