yaqwsx/KiKit #711: V-Cut lines overshoot length, Text layer vs Kicad DRC and 3D view

gwideman, 2024-07-17:
### Prerequisites

- [X] I have read the documentation and the proposed feature is not implemented.

### Description

I see that there has been some previous issues and code revs related to this issue, but I did not see a report of some Kicad behavior this interacts with, in light of which I think one additional feature might be needed.

**Short version:**
- Requesting an option to set the layer for the "V-Cut" text separate from the vcut line layer.

**Rationale**

- Some PCB vendors favor having vcut lines on the same layer as board edge (so, Edge.Cuts). JLC's site says "Please kindly make sure that the V-cut lines, cutouts, millings and slots are in the same layer with the board outline. If it is not in the same layer with the board outline, it will be missed."
- Currently, we can set kikit to do that, but...
- The current generated vcuts have lines extending beyond the board edge, which Kicad flags as DRC errors. This also blocks the 3D viewer from understanding where the board outline is.   I believe this is addressed in 2e25a3d.  Even a correct-length vcut line would bulge beyond the board outline, so vcut thickness needs to be set, which I believe is also addressed in 2e25a3d.
- NEW: in a similar vein, the "V-Cut" text that kikit adds to the board also confuses DRC when it's on Edge.Cuts. (Not known by me  whether it confuses PCB vendor's board software.)  So this calls for ability to set that text to some layer other than Edge.Cuts.  I didn't see this implemented in any commit.

**Related issues I found**
- #652
- #674
- #677 
- 2e25a3d

yaqwsx, 2024-07-17:
If you decide (or your manufacturer forces you) to place V-cuts on Edge.Cuts, you won't have any luck with KiCAD's 3D browser. V-cuts, even if precisely placed on the board edge, will interfere with the closed polygons formed by other lines in the layer that are expected by both KiKit and the 3D viewer.

Separating layers for V-CUT text and line can be done easily (and we will add this); however, as I stated above, I am afraid it won't solve your issue.

I think the proper solution is one of three:
- force your manufacturer to accept V-cuts on a separate layer;
- always generate two panels: one with V-cuts on Edge.Cuts and the other one with some other layer. Use the latter one to preview your panel and generate 3D renders, and send the first one to your manufacturer.
- ensure there is a change in KiCAD such that graphical items have a flag whether they are board-feature significant or just documentation.

gwideman, 2024-07-17:
Thanks for your thoughtful and convincing answer.

closed 2025-01-26 by yaqwsx