yaqwsx/KiKit #714: MacOS installer not working

emm312, 2024-07-17:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

1.5.1

### KiCAD version

8.0.4

### Operating system

MacOS

### Description

I have installed kikit on macos with the script and it runs on the cli, but when I open pcbnew it gives me the 'KiKit backend not found' error. Other issues suggest something is wrong with the install script but I have verified that it uses the fixes mentioned in them.

### Steps to Reproduce

1. Install KiCad via brew
2. Install KiKit via the PCM (I tried reinstalling it after to no avail)
3. Install the KiKit backend with the guide on the website

yaqwsx, 2024-07-19:
Could you try it again with the new 1.6.0 release?

emm312, 2024-07-19:
Oh yeah I forgot to say, I pip install'd the upstream version and it worked perfectly! I needed to use sudo and pass the -U flag

closed 2024-07-19 by emm312