yaqwsx/KiKit #715: win 10 installer not working

mikhailmihalkov, 2024-07-18:
### Prerequisites

- [X] I have read [FAQ](https://yaqwsx.github.io/KiKit/latest/faq/)
- [X] I have searched existing issues (including closed ones)
- [X] I use KiKit at least version 1.5.1 (older version are not supported)

### KiKit version

1.5.1

### KiCAD version

8.0.4

### Operating system

win 10 /64 

### Description

I installed version 1.5.1 using the script, after installation it says that the plugin was not found, although in version 8.0.3 everything worked

### Steps to Reproduce

I installed version 1.5.1 using the script, after installation it says that the plugin was not found, although in version 8.0.3 everything worked

yaqwsx, 2024-07-18:
What script are you referring to? We have no installation script for Windows. These are the official installation instructions for Windows: https://yaqwsx.github.io/KiKit/latest/installation/windows/#installation-on-windows

mikhailmihalkov, 2024-07-18:
> О каком скрипте идет речь? У нас нет скрипта установки для Windows. Вот официальные инструкции по установке для Windows: https://yaqwsx.github.io/KiKit/latest/installation/windows/#installation-on-windows

Yes, everything is installed according to this instruction. The installation is successful, but when launched, a window appears that the plugin is not installed

ZeptoAto, 2024-07-18:
Judging from console output when you type "kikit --version" it has something to do with numpy library being updated to version 2.0.0 but modules still using version 1.x.

yaqwsx, 2024-07-18:
If this is the case, please update to the upstream version - the incompatibility should be fixed. There is a release planned in a few days. There are just a couple of minor bugs I would like to address first.

mikhailmihalkov, 2024-07-18:
> If this is the case, please update to the upstream version - the incompatibility should be fixed. There is a release planned in a few days. There are just a couple of minor bugs I would like to address first.

Please make it possible to update via plugin manager. Thanks!

yaqwsx, 2024-07-18:
** There is a release planned in a few days. There are just a couple of minor bugs I would like to address first.**

yaqwsx, 2024-07-18:
The upstream version fixes this (confirmed in #716). The release will be made today or tomorrow.

closed 2024-07-18 by yaqwsx